qmk/qmk_firmware #2071: Ergodox EZ - can't compule undeclared functions

CMGreenspon, 2017-11-28:
Used the EZ configurator and then wanted to modify it slightly by adding macros but can't compile due to various commands that the online configurator not existing?

`keymaps/keymap_CMG.c:33: error: 'KC_RSPC' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_VAI' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_VAD' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_MOD' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_TOG' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_SLD' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_HUI' undeclared here (not in a function)
keymaps/keymap_CMG.c:75: error: 'RGB_HUD' undeclared here (not in a function)`

The source file doesn't indicate where these functions might be found etc. Any advice would be much appreciated!

drashna, 2017-11-28:
Make sure the file is `/keymaps/CMGreenspon/keymap.c`

And would you mind posting it to pastebin/gist?

CMGreenspon, 2017-11-28:
It was in /keymaps/, moved it to /keymaps/cmg/ but it made no difference.

[Here is a link to the C file.](https://pastebin.com/W4jgpUwB
)

Thank you!

drashna, 2017-11-28:
KC_WWW doesn't exist, or isn't defined (L42). I'm guessing that you mean `KC_WWW_HOME`

On the next line, it needs to be `LGUI(KC_D)', as you still need the "KC_" part. 

On line 51, this should be `KC_AUDIO_MUTE`

For the "VRSN" macro, you need to remove it, or re-add `#include "version.h"` to the very top (that's where these are defined)

I made these changes, and it compiles on my system. 

For reference, the c file and compiled hex:
https://www.dropbox.com/sh/x08vgg92x1rijia/AAAnNgHkyqOvQee4muVbJg7Ba?dl=0

Also, just to make sure, you did this:
https://docs.qmk.fm/getting_started_build_tools.html

CMGreenspon, 2017-11-30:
Sorry was away from work so couldn't attempt this. Worked perfectly though!

Thank you!

closed 2017-11-30 by CMGreenspon