qmk/qmk_firmware #2075: Trouble adding new scancodes

ieure, 2017-11-29:
I need to have Hyper modifiers for Emacs.  Currently, I have my Ergodox set up to send different keycodes (`XF86Forward` / `XF86Back`) for the right/left Hyper modifiers, which are then remapped to the correct symbols with xkb.

What I'd prefer to do is have the keys generate an unused scancode, then map that to the modifiers.  This would prevent confusion in the case that the xkb map isn't working correctly, since the generated scancodes would do nothing.

However, when I try to add them, I don't see them generating any events at all.

These are the QMK changes:

```
diff --git a/tmk_core/common/keycode.h b/tmk_core/common/keycode.h
index a1e0bc9a..5f0da568 100644
--- a/tmk_core/common/keycode.h
+++ b/tmk_core/common/keycode.h
@@ -26,7 +26,7 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 #define IS_ERROR(code)           (KC_ROLL_OVER <= (code) && (code) <= KC_UNDEFINED)
 #define IS_ANY(code)             (KC_A         <= (code) && (code) <= 0xFF)
 #define IS_KEY(code)             (KC_A         <= (code) && (code) <= KC_EXSEL)
-#define IS_MOD(code)             (KC_LCTRL     <= (code) && (code) <= KC_RGUI)
+#define IS_MOD(code)             (KC_LCTRL     <= (code) && (code) <= KC_RHYPER)
 
 
 #define IS_SPECIAL(code)         ((0xA5 <= (code) && (code) <= 0xDF) || (0xE8 <= (code) && (code) <= 0xFF))
@@ -54,6 +54,8 @@ along with this program.  If not, see <http://www.gnu.org/licenses/>.
 #define KC_RCTL KC_RCTRL
 #define KC_LSFT KC_LSHIFT
 #define KC_RSFT KC_RSHIFT
+#define KC_LHYP KC_LHYPER
+#define KC_RHYP KC_RHYPER
 #define KC_ESC  KC_ESCAPE
 #define KC_BSPC KC_BSPACE
 #define KC_ENT  KC_ENTER
@@ -399,6 +401,8 @@ enum hid_keyboard_keypad_usage {
     KC_RSHIFT,
     KC_RALT,
     KC_RGUI,
+    KC_LHYPER,
+    KC_RHYPER
 
     /* NOTE: 0xE8-FF are used for internal special purpose */
 };
```

Using either xinput or xev shows no keypress events are being generated when I press the keys bound to `KC_LHYPER` / `KC_RHYPER`.

Is what I'm trying to do feasible, or do I need to continue appropriating unrelated scancodes?

fauxpark, 2017-12-10:
Generally you should not be editing `keycode.h` directly. In your keymap, you can add a new enum, with the first entry having a value of `SAFE_RANGE`:

```c
enum hyper_keycodes {
  KC_LHYPER = SAFE_RANGE,
  KC_RHYPER
};
```

https://docs.qmk.fm/custom_quantum_functions.html#defining-a-new-keycode

ieure, 2017-12-26:
I tried the suggested approach, but it has the same problem.  When I press the keys with the custom scancodes, no keypress event is visible to the OS.  I tested with both `xev` and `xinput test` -- neither registers any event at all.

I have no idea if this is an issue with the OS (or X keyboard layer) dropping events for scancodes it doesn't know about, or with QMK not sending the event.  Is there some way to determine where the problem lies?

fauxpark, 2017-12-27:
Yeah sorry, the `SAFE_RANGE` method actually doesn't do what you're looking for, as I've since learned. It places the keycode _after_ the internal QMK keycodes, which are _after_ the "real" keycodes, so of course it won't emit anything. You were actually on the right track with your original code above, but instead of being modifiers, they have to be regular keycodes. This is because the data sent to the host contains the state of all eight modifiers in one byte. It's non trivial to add more modifier bits.

Simply tacking them on to the end of `KC_EXSEL` probably won't work if you have media/power keys enabled - `0xA5` is `KC_SYSTEM_POWER`, you'll end up shutting your PC down ;)

Have a look at keycode.h and see if you can find two empty slots, modify IS_KEY to take them into account, and add your keycodes to the switch block in [action_for_key()](https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_common.c#L57).

drashna, 2018-03-25:
Is this still an issue?

fredizzimo, 2018-04-07:
The reason why you can't add keycodes there, is that they would map to the reserved range of the HID specification. So the OS most likely ignores them. Check this document http://www.usb.org/developers/hidpage/Hut1_12v2.pdf, page 53. 

Furthermore, the added keycodes, overlaps with the `internal_special_keycodes`, which QMK uses internally.

etale-cohomology, 2018-05-26:
So how can we add new keycodes or scancodes, then?

The X server admits `248` keycodes (which the max X server keycode, `255`, minus the min X server keycode, `8`, plus `1`). The Linux kernel input event driver, `evdev`, defines keycodes up to `767` (you can find the keycode definitions in `/usr/include/linux/input-event-codes.h`). Yet neither TMK nor QMK make it easy to add *actual* keycodes. Of course this *must* be possible, because the firmware already implements a bunch of keycodes in the first place, so all you need to do is add more. Anyone know how?

-------
@ieure What you're probably looking for is to add new **scancodes**, which is, I think, what the keyboard hardware/firmware actually sends to the kernel (**firmware scancodes** have values like `0x70004`, which is what my keyboard sends with TMK keycode `KC_A`). In the case of Linux, `evdev` and `udev` then take care of mapping these scancodes to **keycodes** and propagating them to userspace, which is what the X server sees, for instance, and which you can see yourself using something like `xev` (my `udev` maps firmware scancode `0x70004` to `evdev` keycode `30` (which is `KEY_A`), and then the X server maps `evdev` keycode `30` to X server keycode `38`; X server keycodes are the same as `evdev` keycodes, but shifted `8`). Then the X server maps X server keycodes to X server **keysyms** (but you can ignore keysyms for most purposes because they add an extra layer of complexity that I've found unnecessary).

To see the actual **firmware scancodes**, you can use `evtest` (must run as `root`, unless your keyboard has permissive read permissions, something you can modify with `udev` but which I don't think is safe because then anyone can keylog your keyboard). Scancodes are *not*, however, an `evdev` event of **type**  `EV_KEY` (unlike **keycodes**), but an `evdev` event of **type** `EV_MSC` with **code** `MSC_SCAN`. (Under `evdev`, an event is a 4-tuple `(time, type, code, value)`.) Then the **value** of this event will be the **firmware scancode**.

For keycodes, what you're looking for is `evdev` events of type `EV_KEY`. Then the **code** is the keycode (whose definition you can find in `/usr/include/input-event-codes.h`).

In order to use all of the keycodes that TMK/QMK define, I've *had* the mess with the `udev` scancode-to-keycode mapping, because many of the TMK keycodes (eg. `KC_ALT_ERASE`, `KC_SYSREQ`, `KC_CANCEL`, `KC_CLEAR`, `KC_PRIOR`, `KC_RETURN`, `KC_SEPARATOR`, `KC_OUT`, `KC_OPER`, `KC_CLEAR_AGAIN`, `KC_CRSEL`, `KC_EXSEL`, to name a few) are not "registered" with `udev` and so they all map to `evdev` keycode 240 (which is `KEY_UNKNOWN`) and to X server keycode `248` (which is the X server's version of `evdev`s `KEY_UNKNOWN`).

Anyways, if someone knows how to get the firmware to add extra scancodes (or keycodes), please do let me know...

fauxpark, 2018-05-27:
First let's take a look at the QMK/TMK HID descriptor, in particular [the section that defines the "basic" keycodes](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/usb_descriptor.c#L78):

```c
HID_RI_USAGE_PAGE(8, 0x07), /* Keyboard */
HID_RI_USAGE_MINIMUM(8, 0x00), /* Reserved (no event indicated) */
HID_RI_USAGE_MAXIMUM(8, 0xFF), /* Keyboard Application */
HID_RI_LOGICAL_MINIMUM(8, 0x00),
HID_RI_LOGICAL_MAXIMUM(16, 0x00FF),
HID_RI_REPORT_COUNT(8, 0x06),
HID_RI_REPORT_SIZE(8, 0x08),
HID_RI_INPUT(8, HID_IOF_DATA | HID_IOF_ARRAY | HID_IOF_ABSOLUTE),
```

This is how HID devices communicate to the host what values it's going to send, and the structure and ranges of them. Here, `USAGE_MINIMUM` and `_MAXIMUM` define the minimum and maximum scancode values, and `REPORT_SIZE` backs this up by declaring each of the six scancodes in the report as being 8 bits long. So, the range of scancodes is in fact only 0 to 255, and this is what is actually being sent over the wire.

Additionally, QMK will only emit the "bare" scancode if it is between `0x04` and `0xA4` (`KC_A` and `KC_EXSEL`):

https://github.com/qmk/qmk_firmware/blob/master/quantum/keymap_common.c#L56:
```c
case KC_A ... KC_EXSEL:
case KC_LCTRL ... KC_RGUI:
    action.code = ACTION_KEY(keycode);
    break;
```

`ACTION_KEY(keycode)` is an alias of `ACTION(ACT_MODS, keycode)`, and is handled by [this chunk of code](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action.c#L217) that pretty much just directly adds the keycode to the report. 

I actually have kind of an example of how to add a new scancode in [this Gist](https://gist.github.com/fauxpark/010dcf5d6377c3a71ac98ce37414c6c4) where I implement the Apple Fn key. Theoretically you could throw out all the messing around with the descriptor and report struct and have the keyboard simply send `0xBD` by removing the `case` statement in `keymap_common.c` and instead placing the keycode above the `ACTION_KEY()`:

```c
case KC_A ... KC_EXSEL:
case KC_APPLE_FN:
case KC_LCTRL ... KC_RGUI:
    action.code = ACTION_KEY(keycode);
    break;
```

You'll also need to add the keycode to the `IS_KEY()` macro in `keycode.h`:

```c
#define IS_KEY(code) (KC_A <= (code) && (code) <= KC_EXSEL) || (KC_APPLE_FN == (code))
```

closed 2018-10-22 by drashna