qmk/qmk_firmware #2081: one shot modifier event occurs after the key to be modified in libinput-1.9.3

tim-tx, 2017-11-30:
I recently upgraded my system to libinput-1.9.3 from libinput-1.9.2. My one shot shift key had worked as expected before the upgrade, but now only works for the bottom row (ZXCVBNM... in qwerty). Hitting the one shot shift and a key on one of the other rows just produces a lowercase character. Xev shows that the shift modifier actually registers after second key for these other rows, but in the correct order for the bottom row. The log is attached, I hit n, m, h, j, OSM(MOD_LSFT) + n, osm + m, osm + h, osm + j. You see the right order for osm + {n,m}, but shift registering second for osm + {h,j}. Downgrading to libinput-1.9.2 fixed the issue. This is on an Ergodox EZ, my layout is [here](https://github.com/tim-tx/qmk_firmware/blob/a7e5f6697d7c487e5d2c03d061d6bd03e50e14e4/keyboards/ergodox_ez/keymaps/tsj/keymap.c).

[xev.log](https://github.com/qmk/qmk_firmware/files/1518625/xev.log)

pushcx, 2017-12-04:
I also have this bug on a Planck using a custom layout - same behavior that keys on the same row as the one-shot key work but not others.

drashna, 2018-03-25:
This should work in `libinput 1.9.4-1` now.  Closing. 

If it doesn't, please reopen

closed 2018-03-25 by drashna