qmk/qmk_firmware #2086: Issue building ergodox firmware

DwordPtr, 2017-12-02:
Hi I'm getting the following error message when I try to build the ergodox firmware for either hand.
To recreate this issue:

1. run `make git-submodule`
2. copy-paste one of the ergodox docker commands in the main project folder.

btw I'm running the latest code from master.

````

QMK Firmware 0.5.179
Makefile:528: keyboards/ergodox/rules.mk: No such file or directory
Making ergodox with keymap narze and target clean

lib/lufa/LUFA/Build/LUFA/lufa-sources.mk:29: *** Makefile ARCH option cannot be blank.  Stop.
make: *** [clean] Error 1
Make finished with errors
Makefile:528: recipe for target 'clean' failed
QMK Firmware 0.5.179
Makefile:528: keyboards/ergodox/rules.mk: No such file or directory
Making ergodox with keymap narze and target ergodox:narze

lib/lufa/LUFA/Build/LUFA/lufa-sources.mk:29: *** Makefile ARCH option cannot be blank.  Stop.
make: *** [ergodox:narze] Error 1
Makefile:528: recipe for target 'ergodox:narze' failed
Make finished with errors
zsh: command not found: avr-objcopy
zsh: command not found: dfu-util

```

trunneml, 2017-12-02:
What's the exact docker command you are running? I had to add the make command to the docker container like this:

```sh
docker run --rm -v ~/qmk_firmware:/qmk:rw -it edasque/qmk_firmware make ergodox_ez:default
```

DwordPtr, 2017-12-03:
I have been running 
````docker run -e keymap=narze -e subproject=infinity -e keyboard=ergodox --rm -v $('pwd'):/qmk:rw edasque/qmk_firmware
avr-objcopy -Iihex -Obinary .build/ergodox_infinity_narze.hex .build/ergodox_infinity_narze_left.bin
dfu-util --device 1c11:b007 -D .build/ergodox_infinity_narze_left.bin
```

It tells me lib lufa can't compile because the `ARCH` options is blank?
I'm new to this and I have no idea what it means.

DwordPtr, 2017-12-11:
this worked for me by skipping docker. If anybody is having issues try `sudo make:egrodox_infinity:<layout>:dfu-util` I'll take a look and see if I can get the docker container working sometime eventually.

closed 2017-12-11 by DwordPtr

MarioAriasC, 2018-03-26:
I still have the same problem and my only way to compile it is using docker.

Did you find an answer?

DwordPtr, 2018-03-27:
Unfortunately no. I just built it from my local os.