qmk/qmk_firmware #2087: keymap fails to compile with MOUSEKEY_ENABLE=yes

mattfair, 2017-12-03:
I have a keymap (planck rev4) that compiles for fine in release 0.5.174, but breaks in releases 0.5.175+.  The error I get is:

```
Compiling: keyboards/planck/planck.cIn file included from ./tmk_core/common/report.h:83:0,
                 from ./quantum/keymap.h:29,
                 from ./quantum/quantum.h:26,
                 from keyboards/planck/planck.h:4,
                 from keyboards/planck/planck.c:1:
./tmk_core/protocol/lufa/descriptor.h:246:3: error: #error "Endpoints are not available enough to support all functions. Remove some in Makefile.(MOUSEKEY, EXTRAKEY, CONSOLE, NKRO, MIDI, SERIAL)"
 # error "Endpoints are not available enough to support all functions. Remove some in Makefile.(MOUSEKEY, EXTRAKEY, CONSOLE, NKRO, MIDI, SERIAL)"
   ^
 [ERRORS]
```
My rules.mk does have:
```
MOUSEKEY_ENABLE=yes
```

If I set in my rules.mk:
```
MOUSEKEY_ENABLE=no
```
It will then build successfully.  Is there something I need to add to to make my code compile with the newer releases?  Using the `:production` make flag doesn't work either.

drashna, 2017-12-03:
Tested this out. Add `MIDI_ENABLE=no`, and that should fix the issue (at least, it did for me).

mattfair, 2017-12-03:
yep that was it, thanks!

closed 2017-12-03 by mattfair