qmk/qmk_firmware #2089: undefined reference to `_sbrk' when using AutoShift with Ergodox Infinity

twhiston, 2017-12-03:
Hi,
I found that if i enabled AutoShift on the Ergodox Infinity the build failed.
```
Linking: .build/ergodox_infinity_workman.elf                                                        [ERRORS]
 | 
 | /usr/local/Cellar/gcc-arm-none-eabi/20160928/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/libg.a(lib_a-sbrkr.o): In function `_sbrk_r':
 | sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [.build/ergodox_infinity_workman.elf] Error 1
make: *** [ergodox_infinity:workman:dfu-util] Error 1
```

Solution seems to be to add `$(CHIBIOS)/os/various/syscalls.c` to the SRC value in rules.mk for the board, but not sure if you want that always added.

drashna, 2018-03-25:
If you alter the make process, you don't need to always add it.  

I've issued a pull request that fixes this: #2608.

You can pull the diff from that page and apply. But I can confirm that it compiles properly now.

fredizzimo, 2018-04-07:
The real reason why it fails, is that it's including `stdio.h` and using `snprintf`.  The QMK print functions should be used instead of that, since we don't have a standard console for the output. 

I'm re-opening this issue, so that a proper fix can be made. I will probably do that tomorrow.

drashna, 2018-10-22:
This should be fixed in #2608 I think.

jackhumbert, 2020-03-17:
I think this is fixed now with #8268 being merged - if not, please reopen!

closed 2020-03-17 by jackhumbert