qmk/qmk_firmware #2098: Proposals: support "default" and "after" processing behaviors, and add helper function `get_record_keycode`.

colinta, 2017-12-04:
# Proposal 1: `process_record_system`

In my keymap I'm setting LED based on the state of layers, and in `process_record_user` I wanted to use the "default" behavior, but then query the state of the modifiers (for a custom "sticky key" implementation, but that's not relevant).

My solve was to refactor `process_record(keyrecord_t *)` and move the "default/system" record handling into a separate function.  This change is very easy:

```diff
diff --git a/tmk_core/common/action.c b/tmk_core/common/action.c
index b39aa4c..70a8b83 100644
--- a/tmk_core/common/action.c
+++ b/tmk_core/common/action.c
 void process_record(keyrecord_t *record)
 {
     if (IS_NOEVENT(record->event)) { return; }
@@ -139,6 +143,11 @@ void process_record(keyrecord_t *record)
     if(!process_record_quantum(record))
         return;

+    process_record_system(record);
+}
+
+void process_record_system(keyrecord_t *record) {
     action_t action = store_or_get_action(record->event.pressed, record->event.key);
     dprint("ACTION: "); debug_action(action);
 #ifndef NO_ACTION_LAYER
```

With this change, in my `process_record_user` function I could call out to the `_system` implementation and then perform some cleanup / state checking, and then `return false;` to indicate I handled everything.

# Proposal 2: `process_record_after`

This is not strictly necessary, because it could be functionally similar to using `process_record_system`, but 
in my case I wanted an all purpose "any time the layer changes, set the LED" logic.  Finding all the places the layer changes was tedious, but using `process_record_after` made it easy.

This change piggy backs on `process_record_system` change:

```diff
diff --git a/tmk_core/common/action.c b/tmk_core/common/action.c
index b39aa4c..70a8b83 100644
--- a/tmk_core/common/action.c
+++ b/tmk_core/common/action.c
@@ -132,6 +132,10 @@ bool process_record_quantum(keyrecord_t *record) {
     return true;
 }

+__attribute__ ((weak))
+void process_record_after(keyrecord_t *record) {
+}
+
 void process_record(keyrecord_t *record)
 {
     if (IS_NOEVENT(record->event)) { return; }
@@ -139,6 +143,11 @@ void process_record(keyrecord_t *record)
     if(!process_record_quantum(record))
         return;

+    process_record_system(record);
+    process_record_after(record);
+}
+
+void process_record_system(keyrecord_t *record) {
     action_t action = store_or_get_action(record->event.pressed, record->event.key);
     dprint("ACTION: "); debug_action(action);
 #ifndef NO_ACTION_LAYER
```

# Proposal 3: `get_record_keycode`

In using `process_record_after` I immediately realized that I needed the `uint16_t keycode`, so I refactored `process_record_quantum` to pull that logic into a helper function.  This just felt like a great refactor anyway, that's a pretty big function and cleaning it up felt good.

```diff
--- a/quantum/quantum.c
+++ b/quantum/quantum.c
@@ -168,26 +168,7 @@ static uint16_t scs_timer[2] = {0, 0};
 static bool grave_esc_was_shifted = false;

 bool process_record_quantum(keyrecord_t *record) {
-
-  /* This gets the keycode from the key pressed */
-  keypos_t key = record->event.key;
-  uint16_t keycode;
-
-  #if !defined(NO_ACTION_LAYER) && defined(PREVENT_STUCK_MODIFIERS)
-    /* TODO: Use store_or_get_action() or a similar function. */
-    if (!disable_action_cache) {
-      uint8_t layer;
-
-      if (record->event.pressed) {
-        layer = layer_switch_get_layer(key);
-        update_source_layers_cache(key, layer);
-      } else {
-        layer = read_source_layers_cache(key);
-      }
-      keycode = keymap_key_to_keycode(layer, key);
-    } else
-  #endif
-    keycode = keymap_key_to_keycode(layer_switch_get_layer(key), key);
+  uint16_t keycode = get_record_keycode(record);

   // This is how you use actions here
   // if (keycode == KC_LEAD) {
@@ -603,6 +584,27 @@ bool process_record_quantum(keyrecord_t *record) {
   return process_action_kb(record);
 }

+uint16_t get_record_keycode(keyrecord_t *record) {
+  /* This gets the keycode from the key pressed */
+  keypos_t key = record->event.key;
+
+  #if !defined(NO_ACTION_LAYER) && defined(PREVENT_STUCK_MODIFIERS)
+    /* TODO: Use store_or_get_action() or a similar function. */
+    if (!disable_action_cache) {
+      uint8_t layer;
+
+      if (record->event.pressed) {
+        layer = layer_switch_get_layer(key);
+        update_source_layers_cache(key, layer);
+      } else {
+        layer = read_source_layers_cache(key);
+      }
+      return keymap_key_to_keycode(layer, key);
+    } else
+  #endif
+    return keymap_key_to_keycode(layer_switch_get_layer(key), key);
+}
```

Obviously things like naming are up to core team, these were my attempt at consistency, but Jack et. al. will know better.  Are these good changes / refactors?

colinta, 2017-12-04:
Two questions brought up in gitter.im:
- is this similar to #1525?  No, because that branch is about changing the `return true/false;` "magic" return values to more intuitive `STOP_PROCESSING/CONTINUE_PROCESSING` names, which I also would love to see.  These proposals are about having places to put code that goes *after* key processing.
- can it be accomplished by using `!record->event.pressed` logic?  It can, that's where most "after key press" code goes, but in my case I wanted to have "after layer_state change" logic, and that isn't always 1::1 with key-up events.  Also, if I changed my layer macros from `TT()` to `TO()`, my state change code would be wrong again.  Instead, if I use a generic `_after()` function, I can just check the value of `layer_state` and act accordingly.  Similarly for things like changes to modifier keys.

colinta, 2017-12-04:
- `layer_state_set_user` could def be used for the layer examples.  There isn't a similar callback for modifier key changes (I'm setting LEDs based on modifier "sticky" setting)

TurboMech, 2017-12-07:
@colinta I’m working on a similar issue. See https://github.com/qmk/qmk_firmware/issues/2109. The layer RGB is working if you set it as a key code.  The issue I’m having is after reset the keyboard will return to the last RGB mode but as soon as you toggle a layer (or in the code I posted CAPS LOCK) and then return to default layer, the RGB mode is “forgotten” and reverts to mode 1 (static), but the hue will remain. 

So using your code would this persist after a reboot and then after a layer change?

drashna, 2017-12-07:
@TurboMech If you're using `_setrgb`, then that would be why.  IIRC, it manually sets the mode to "1" (solid).  
Using `_sethsv` works better, keeps the mode or lets you set a mode, as well.

fredizzimo, 2018-04-07:
I think that for this use case, `layer_state_set_user` should be used, and that we should add a similar callback for modifiers too.

For the general case I would like to see something like 1., but extended a bit further, so that can emit any keycode from `process_record_user`, basically like `register_code16`, but not limited to keycodes that are sent directly. This would probably have a lot of different use cases.

That could be implemented by having a queue at the core, so when you add a new keycode it's added to the queue, which is then processed before any new physical keypresses. The reason for using the queue to prevent too much recursion in our code base. 

Key records injected this way could have a flag, maybe `user`, so that you can ignore the second time it's  processed, if you do like your original post suggest. 

With this approach you are also not limited to injecting just one keycode, you could for example queue up many codes, to implement some very advanced macros, which not only sends physicsl key presses but also controls QMK in itself for example midi codes and rgb codes.

colinta, 2018-04-07:
First: very much in agreement on the "event queue" that you described, I'm gonna touch on your first comment, about using `layer_state_set_user`

Every time I've discussed this PR (in gitter, though it's been a while), there has been an unfortunate focus on the example I gave, which is related to layers, and so yes logically that could go in `layer_state_set_user`.  So let's move past that example and think about other scenarios where "after processing" might be an appropriate place to put some code:

- an LED that is either on, off, or blinking, and that state is based on a combination of modifier key and whether a macro is being recording
- listening to key chord events, where the order the keys are pressed doesn't matter
- starting/stopping a "most recent activity" timer

So it's very true that all these examples *could* be put someplace else, but in my local fork I have this `after` event and it is a *more logical* place to put the logic for events that happen regardless of the specific key that was pressed; I have lots of `process_record_blabla` functions that can handle specifc keys, and many overlap on changing modifier keys or internal state, so having a guaranteed "this function is called *after* all that nonsense" is really handy.  That's all I'm saying. :smile:

colinta, 2018-04-09:
For the curious, here are the commits where I made these changes (I recently rebased them onto master, a PR for any/all of these would be easy)

`get_record_keycode` -> https://github.com/colinta/qmk_firmware/commit/3ce1567be2ebd7622b7b73586e56fd22d768b7b4
`process_record_system` -> https://github.com/colinta/qmk_firmware/commit/fd99c65dcf9cec575a84e58d87016bdd14ba0ec2
`process_record_after` -> https://github.com/colinta/qmk_firmware/commit/0c2484d33c611a6825c79d87b19bb44a6530fe21

PR #4892 (2020-03-22): Add Post Processing to process_record
## Description
This adds `post_process_record_*` in addition to `process_record_*`, so that you can run cleanup/handling AFTER a keystroke has been sent. 

This code was developed by @colinta, and pursued by myself after conversations on Discord.

This also moves the "report to keycode" code to a separate function, that can be called elsewhere, since it is being used by both `process_record_quantum()` and `post_process_record_quantum()` now. (`get_record_keycode` and `get_event_keycode`)

Also, I updated the formatting of the touched files to follow the Co [... cut]
closed 2020-03-22 by tzarc