qmk/qmk_firmware #2099: Error when flashing vea AVRDUDE in MSYS

drashna, 2017-12-04:
When using MSYS2 on Windows, and using `:avrdude`, it generates this error:

```
Detected controller on USB port at /dev/ttyS4
avrdude.exe: ser_open(): can't open device "/??/COM5": The system cannot find the path specified.

avrdude.exe done.  Thank you.

make[1]: *** [tmk_core/avr.mk:172: avrdude] Error 1
Make finished with errors
```

It looks like it's outputting the wrong string (`/??/COM` rather than just `COM`)

Tiryoh, 2018-02-13:
I don't have the Keyboard yet, but I think editing `tmk_core/avr.mk` in line 184-187 will solve the problem.
Would you mind trying this?

edit this file(`tmk_core/avr.mk`)
https://github.com/qmk/qmk_firmware/blob/c5d81a84a0519c78de1b5e165f5f2c1eb773bc11/tmk_core/avr.mk#L184-187
```
		echo ""; \
		echo "Detected controller on USB port at $$USB"; \
		sleep 1; \
		avrdude -p $(MCU) -c avr109 -P $$USB -U flash:w:$(BUILD_DIR)/$(TARGET).hex; \
```

as this:
```
		echo ""; \
		if grep -q -s MSYS /proc/version; then \
			USB=`echo $$USB | sed -e 's/\/dev\/ttyS\([0-9]*\)/\1/g' | awk '{print "COM"$1+1}'`; \
		fi; \
		echo "Detected controller on USB port at $$USB"; \
		sleep 1; \
		avrdude -p $(MCU) -c avr109 -P $$USB -U flash:w:$(BUILD_DIR)/$(TARGET).hex; \
```

Please be careful that you should indent with TABs in Makefile.

Tiryoh, 2018-02-14:
Thanks to [pluis9](https://twitter.com/pluis9/status/963803732889632768), I think I found the way to fix this issue.
We need to edit this file(`tmk_core/avr.mk`)
https://github.com/qmk/qmk_firmware/blob/c5d81a84a0519c78de1b5e165f5f2c1eb773bc11/tmk_core/avr.mk#L184-187
as this:
```
		echo ""; \
		if grep -q -s MINGW64 /proc/version; then \
			USB=`echo $$USB | sed -e 's/\/dev\/ttyS\([0-9]*\)/\1/g' | awk '{print "COM" ($1+1)}'`; \
		fi; \
		echo "Detected controller on USB port at $$USB"; \
		sleep 1; \
		avrdude -p $(MCU) -c avr109 -P $$USB -U flash:w:$(BUILD_DIR)/$(TARGET).hex; \
```

[image]

Tiryoh, 2018-02-14:
Thanks to [pluis9](https://twitter.com/pluis9/status/963803732889632768), I think I found the way to fix this issue.
We need to edit this file(`tmk_core/avr.mk`)
https://github.com/qmk/qmk_firmware/blob/c5d81a84a0519c78de1b5e165f5f2c1eb773bc11/tmk_core/avr.mk#L184-187
as this:
```
		echo ""; \
		if grep -q -s MINGW64 /proc/version; then \
			USB=`echo "COM"$$(($$(echo $$USB | sed -e 's/\/dev\/ttyS\([0-9]*\)/\1/g') + 1))`; \
		fi; \
		echo "Detected controller on USB port at $$USB"; \
		sleep 1; \
		avrdude -p $(MCU) -c avr109 -P $$USB -U flash:w:$(BUILD_DIR)/$(TARGET).hex; \
```
[image]

drashna, 2018-02-14:
Nope, doesn't work here.  
And I suspect that you  have a very out of date version.

Tiryoh, 2018-02-14:
@drashna,

Thanks for trying.

> Nope, doesn't work here.

Do you mean that you got the same error like https://github.com/qmk/qmk_firmware/issues/2099#issue-279117518?

> And are you on an older version?

I'm sorry to have confused you.  I'm working on this version : b89e318d35b295465a14679f27d71077a51daa3b
I specified c5d81a84a0519c78de1b5e165f5f2c1eb773bc11 for the URL because this commit is the latest version of `tmk_core/avr.mk`.

drashna, 2018-02-14:
Ah, doesn't help that I'm using a hacky hack for running MSYS2 (in VS2017).  The second one works.  I have to change it to MSYS because that's how it shows up in VS2017.

Tiryoh, 2018-02-14:
@drashna,

Thanks for the information.
I think this might work.
```
		echo ""; \
		if grep -q -s MINGW /proc/version || grep -q -s MSYS /proc/version; then \
			USB=`echo "COM"$$(($$(echo $$USB | sed -e 's/\/dev\/ttyS\([0-9]*\)/\1/g') + 1))`; \
		fi; \
		echo "Detected controller on USB port at $$USB"; \
		sleep 1; \
		avrdude -p $(MCU) -c avr109 -P $$USB -U flash:w:$(BUILD_DIR)/$(TARGET).hex; \
```

drashna, 2018-02-14:
Success!!  This work in MSYS, MINGW, and in my hacky whack-whack terminal session!

drashna, 2018-02-14:
Checking another in progress PR... this looks cleaner: 

`if grep -q -s 'MINGW\|MSYS' /proc/version; then \`

So, it would be: 
```
		echo ""; \
			if grep -q -s 'MINGW\|MSYS' /proc/version; then \
			USB=`echo "COM"$$(($$(echo $$USB | sed -e 's/\/dev\/ttyS\([0-9]*\)/\1/g') + 1))`; \
    fi; \
		echo "Detected controller on USB port at $$USB"; \
		sleep 1; \
		avrdude -p $(MCU) -c avr109 -P $$USB -U flash:w:$(BUILD_DIR)/$(TARGET).hex; \
```

closed 2018-02-14 by drashna