qmk/qmk_firmware #2111: OSM / One shot failing to trigger

sQVe, 2017-12-06:
I have a **Ergodox EZ** keyboard and my keymap has previously has **OSM** keys set and it has been working flawlessly. Since I updated my keymap that has changed. My settings and keymap can be found at: https://github.com/sQVe/qmk_firmware/tree/sqve/keyboards/ergodox_ez/keymaps/sqve

Am I the only one having this issue? Would love some help in solving this. :beers:

RaeedChowdhury, 2017-12-08:
I'm having a similar issue with my keyboard, where some of my one shot modifiers don't work. I think, though, that I narrowed it down to something going on in my i3 window manager, and nothing to do with my keyboard, since my OSM keys work fine when I exit i3 and also when I switch to Windows. I'm not sure how to fix it, though. Any chance you're also using i3 window manager on Linux?

sQVe, 2017-12-08:
Nicely spotted @RaeedChowdhury! I'm indeed also using i3wm. I'll do some downgrading later to see where this issue was introduced. I'm a bit unsure how we, in the best way, explain this issue in the i3wm repo.

sQVe, 2017-12-08:
@RaeedChowdhury I've tried downgrading `i3-wm` down three versions and it made no difference. I also tried downgrading all its dependencies and no difference. I think there is something else at fault here...

RaeedChowdhury, 2017-12-08:
Actually, I did some digging last night into what keycodes are being read in, and it seems like it has something to do with how Xorg is reading in the keyboard inputs. I'm not sure if this is happening to you, but strangely, my OSM left shift works only for a few letters, like 'z' and 'x', while my OSM left ctrl works for most of my keyboard. The other OSMs appear to not be working.

The reason, as far as I can tell, is that when the OSM triggers, the keyboard sends the modifier code and the other pressed key code at almost the same time, and the order that Xorg sees the keycodes gets flipped sometimes now. The reason for this, it seems, is more subtle. Looking deeper, it seems that whatever reads keyboard input in Xorg reads the keycodes in ascending order, if the keycodes come close enough in time. For example, the keycode for left shift is 42, and most of the alpha keys on the keyboard have keycodes less than 42. But 'z' has keycode 44, so it works with left shift.

I don't remember what package reads keyboard input in Xorg, but an update to that package appears to be the problem for me, though I haven't downgraded to check. I'm not sure how to fix it, other than change the QMK code to actually send and hold the OSM keycode until the next key is pressed or for the entire timeout length. Incidentally, this is how the documentation says that One Shot Modifiers work, so it's probably a reasonable change to try to implement.

sQVe, 2017-12-08:
@RaeedChowdhury I agree. If this is the case it should really be fixed here.

RaeedChowdhury, 2017-12-08:
I'm happy to try my hand at a fix but I don't know QMK or TMK well enough to even know how to implement this behavior. @jackhumbert Do you have any suggestions on where to start (or even whether this is advisable)?

RaeedChowdhury, 2017-12-09:
A little bit of a kludge for now though: I added a send_keyboard_report() line at the end of set_oneshot_mods() and clear_oneshot_mods() in tmk_core/common/action_util.c, and it seems to have the desired effect for my test case. I'm pretty sure that this isn't the ideal fix though, and it probably doesn't address the ONESHOT_TAP_TOGGLE functionality.

RaeedChowdhury, 2017-12-11:
...Actually, that seems to reverse the problem, where in some cases, Xorg is registering the modifier release before the keypress. I think the fix is going to have to be more nuanced.

j-kramer, 2017-12-11:
@RaeedChowdhury your description of the bug matches with [this bug](https://bugs.freedesktop.org/show_bug.cgi?id=104030) in libinput. Downgrading libinput should fix the issue as described in issue #2081.

RaeedChowdhury, 2017-12-12:
Thanks, I agree, you're probably right. I think I'll just take one-shot modifiers out of my keymap until someone decides to patch libinput.

Flink, 2017-12-13:
I had exactly the same problem as you people and downgrading libinput did fix the bug indeed 👍

sQVe, 2018-01-29:
This now works with `libinput 1.9.4-1`. Closing.

closed 2018-01-29 by sQVe