qmk/qmk_firmware #2122: make shows grep usage on macOS. grep -P is not supported

nachie, 2017-12-08:
⌘ make mitosis:default
QMK Firmware 0.5.184
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 make git-submodule
 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself.

Making mitosis with keymap default

avr-gcc (GCC) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/mitosis/matrix.c                                                               [OK]
Compiling: keyboards/mitosis/mitosis.c                                                              [OK]
Compiling: keyboards/mitosis/keymaps/default/keymap.c                                               [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/process_keycode/process_leader.c                                                 [OK]
Compiling: ./tmk_core/common/host.c                                                                 [OK]
Compiling: ./tmk_core/common/keyboard.c                                                             [OK]
Compiling: ./tmk_core/common/action.c                                                               [OK]
Compiling: ./tmk_core/common/action_tapping.c                                                       [OK]
Compiling: ./tmk_core/common/action_macro.c                                                         [OK]
Compiling: ./tmk_core/common/action_layer.c                                                         [OK]
Compiling: ./tmk_core/common/action_util.c                                                          [OK]
Compiling: ./tmk_core/common/print.c                                                                [OK]
Compiling: ./tmk_core/common/debug.c                                                                [OK]
Compiling: ./tmk_core/common/util.c                                                                 [OK]
Compiling: ./tmk_core/common/eeconfig.c                                                             [OK]
Compiling: ./tmk_core/common/report.c                                                               [OK]
Compiling: ./tmk_core/common/avr/suspend.c                                                          [OK]
Compiling: ./tmk_core/common/avr/timer.c                                                            [OK]
Compiling: ./tmk_core/common/avr/bootloader.c                                                       [OK]
Assembling: ./tmk_core/common/avr/xprintf.S                                                         [OK]
Compiling: ./tmk_core/common/magic.c                                                                [OK]
Compiling: ./tmk_core/common/mousekey.c                                                             [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Compiling: ./tmk_core/protocol/lufa/lufa.c                                                          [OK]
Compiling: ./tmk_core/protocol/lufa/descriptor.c                                                    [OK]
Compiling: ./tmk_core/protocol/lufa/outputselect.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Linking: .build/mitosis_default.elf                                                                 [OK]
Creating load file for flashing: .build/mitosis_default.hex                                         [OK]
Copying mitosis_default.hex to qmk_firmware folder                                                  [OK]
usage: grep [-abcDEFGHhIiJLlmnOoqRSsUVvwxZ] [-A num] [-B num] [-C[num]]
	[-e pattern] [-f file] [--binary-files=value] [--color=when]
	[--context[=num]] [--directories=action] [--label] [--line-buffered]
	[--null] [pattern] [file ...]

jackhumbert, 2017-12-11:
cc @tengg 

Confirmed - sorry about that. Luckily it's only the file size comparison that's broken here. I think ack can be used here instead, right? We use that in other places.

The offending command is ` grep -oP "(?<=AVR_SIZE: ).+"`` located [here](https://github.com/qmk/qmk_firmware/commit/9fdc27626097ae03b767a09427efc90475d90955#diff-a411ea609aa598b217af5b21e90e5094R374)

yhara, 2018-01-21:
(FYI) Here is a workaround:

```
$ brew tap homebrew/dupes
$ brew install grep --with-default-names   # This will install /usr/local/bin/grep
$ PATH=/usr/local/bin:$PATH make lets_split/rev2:default
```

rickard-von-essen, 2018-01-21:
@yhara that has dangerous implications.

rickard-von-essen, 2018-01-21:
> Every grep

There are actually only two. But yes the options are a) support both or b) use `ggrep` and require that BSD's install it.

PR #2299 (2018-02-02): Use perl directly instead of grep -P
Fixes #2122
closed 2018-02-02 by jackhumbert