qmk/qmk_firmware #2129: Send custom key-codes using tap-dance

danielo515, 2017-12-10:
Hello,

I'm building a macropad and I'm really enjoying using QMK. I really love the tap dance feature, expands the keyboard usability greatly. I was wondering if it is possible to send custom key-codes using tap dance macros. I have two custom keycodes and I want to send them based on the number of taps. I tried the following with no luck:

```
enum custom_keycodes
{
   T_TERM = SAFE_RANGE,
   BLK_COMMENT,
   LINE_COMMENT,
 };

  qk_tap_dance_action_t tap_dance_actions[] = {
    [COMMENT] = ACTION_TAP_DANCE_DOUBLE(BLK_COMMENT, LINE_COMMENT)
  };
```
It compiles, but the code on process_record_user is never executed. I know my macros are correct because they work when they are not invoked using tap dance

thanks in advance

drashna, 2017-12-10:
I'd have to check the code, but I suspect that it's doing something "weird" here, and that's why it's not being processed. 

However, the first example ***"should"*** work: https://docs.qmk.fm/feature_tap_dance.html#complex-examples

danielo515, 2017-12-10:
hELLO @drashna,

Thanks for your prompt response. Trying what you suggested gives me compile errors:

```
In function 'dance_comment_finished':
keyboards/danielomp/keymaps/default/keymap.c:78:5: error: large integer implicitly truncated to unsigned type [-Werror=overflow]
```

Looks like register_code doesn't like my custom key-codes:

```
void dance_comment_finished (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
    register_code (LINE_COMMENT);
  } else {
    register_code (BLK_COMMENT);
  }
}

void dance_comment_reset (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
    unregister_code (LINE_COMMENT);
  } else {
    unregister_code (BLK_COMMENT);
  }
}

qk_tap_dance_action_t tap_dance_actions[] = {
    [ENT_5] = ACTION_TAP_DANCE_DOUBLE(KC_5, KC_ENT),
    [ZERO_1] = ACTION_TAP_DANCE_DOUBLE(KC_1, KC_0),
    [COMMENT] = ACTION_TAP_DANCE_FN_ADVANCED (NULL, dance_comment_finished, dance_comment_reset)
  };
```

drashna, 2017-12-10:
Rather than registering the keycodes, call the code here.  

Maybe add it to a custom function, so you don't have duplicate code.

danielo515, 2017-12-10:
I already have a custom function because I have several similar macros. I'll do that. Can I just call SEND_STRING and that stuff normally within a tap dance ?

drashna, 2017-12-10:
Yes, that would work, too.

danielo515, 2017-12-10:
Thanks for the tip @drashna your proposed workaround works.

However, I would prefer to be able to just send the key-codes normally and let the firmware handle them like the rest of my macros where no tap dance is involved. But I can wait for that, a bit of duplication (even using a function) does not hurt so much at this moment

Thanks a lot!

drashna, 2017-12-10:
Yes, absolutely.  And without digging, I'm not sure why. I can make some guesses, but basically, the custom codes are added "too late" to properly process these.

drashna, 2018-03-25:
Did you get this resolved/working?

danielo515, 2018-03-25:
Nope. This is still an issue. It happens both on Atmega-32U4 and Atmega-32A

JacobJerrell, 2018-04-06:
I think I know why this is happening. I can't be sure without seeing the definition of the custom keycode but I just ran into the same error when trying to setup a tap dance routine.

You cannot use shifted keycodes in tap dance functions. For example, I was trying to use `KC_TILDE` in mine. I fixed the problem by registering KC_LSHIFT + KC_GRAVE and then unregistering KC_GRAVE + KC_LSHIFT to produce the tilde character (I don't know if the order matters on the unregister but it seemed logical). For another example, if you wanted to add a parenthesis then you would want to register and unregister the shift keycode and the '9' keycode.

Hope this helps you

drashna, 2018-10-22:
Unfortunately, there is no way to send custom keycodes here, properly.

The two real options to do this are: 
1. create a custom function and call that from both the `process_record_user` function, and from a custom tap dance
2. use a custom tapdance, create the "record" structure, and call `process_record_user` directly.

danielo515, 2018-10-24:
Should this be added to the FAQ?

drashna, 2018-10-24:
Which part?

danielo515, 2018-10-25:
The part that specifically says that sending custom keycodes with tap dance is not possible

drashna, 2018-10-25:
Ah, probably. 

And yeah, it only supports "basic" codes, since it's calling `register_code16`

danielo515, 2018-12-20:
I think this can be closed, I just checked the docs of tap dance and I saw this:

[image]

closed 2018-12-20 by danielo515

SalchiPapa, 2018-12-20:
I can send shifted key codes in tap dances. Are you using the advanced function or the double tap helper?

danielo515, 2018-12-21:
Hello @SalchiPapa ,
The  idea was about using the regular tap dance macros in order to send custom keycodes. Then those key codes should be handled on the regular eqy (for example on the user function for handling keycodes ). Obviously using the function method it will work, but the idea was to avoid writing functions at all

jared-mackey, 2021-01-07:
This bit me in the butt for quite awhile. :( 

Adding to the docs that tap dances can't send custom keycodes would've saved me a lot of time trying to debug this. Luckily I found this thread and was able to get my problem solved quickly. :)