qmk/qmk_firmware #2130: M() does not work as parameter of LT([layer], M([...]))

Pawnerd, 2017-12-10:
As the title says; It's impossible to use a macro as second parameter in the LT() keycode/command.

hiro2016, 2017-12-13:
That is so by the design of KEYMAP macro I believe.

Qmk relies on tmk core if I'm not mistaken and tmk design dictates:  
1.  firmware detects keypresses and their positions.  
2.  firmware translates a keypress into an action code based on the current layer. 
3. An action code is 16 bits long int.
4. action code has an hid usage id in lower 8 bits and tmk specific code in upper 8bits.    
  
Now LT is a tmk or qmk internal code that goes into upper 8 bits of action code.
Macro is also a tmk or qmk internal code that goes into upper 8bits of action code.  
Two codes cannot be in same action code as adding the two turns them into a different code.  
  
I'm not exactly happy with the current design neither.   
It should be fixed but the problem is who is going to do that... hassu should?  jackhumbert should? or someone else or you or me?    
Personally I think inserting an extra function after tmk&qmk internal processing is done and before the keycode is sent to the PC to allow using hid usage ids as macro triggers on some user defined layers is the simplest hack;allow users to re-interpret the id  in that inserted function, so on some layers sending id for print_screen, a etc triggers a macro. Although I never know if it is really easy till I try it.

drashna, 2018-03-25:
And IIRC, it doesn't support any of the special cases/codes either. It's limited to normal keys. 

This is a limitation for a LOT of the dual function stuff, and something that maybe needs to be fixed in the future.
However, doing so may increase the size of the firmware. 

Additionally, closing for now, as this issue is commonly brought up.

closed 2018-03-25 by drashna