qmk/qmk_firmware #2143: Update Input Club keyboards to use I:C Vendor ID and matching device IDs

belak, 2017-12-12:
Relevant conversation is in the following comments:

- https://github.com/qmk/qmk_firmware/pull/1963#discussion_r149540325
- https://github.com/qmk/qmk_firmware/pull/1963#issuecomment-342670875
- https://github.com/qmk/qmk_firmware/pull/1963#issuecomment-342842597
- https://github.com/qmk/qmk_firmware/pull/1963#issuecomment-342919138

Essentially, this will allow even keyboards flashed with QMK to show up in the configurator.

tengg, 2017-12-15:
Hi,
Just noticed the k-type support has been checked in.
How to flash it? Using what target?

Thanks

tengg, 2017-12-15:
nvm, found it's dfu-util.

Just wondering if it works with RGB lighting?

belak, 2017-12-15:
The QMK port for the K-Type doesn't support LEDs at all yet. Support needs to be written for the chip which controls the lighting and I'm not as familiar with how that works.

tengg, 2017-12-16:
Thanks.
Another issue is the boot magic jump to bootloader seems not working.
I didn't see the bootloader size set in rules.mk, is it supposed to set some value or this chipset is different?

drashna, 2017-12-16:
Looks like it's missing a "`BOOTLOADER = `" line is missing from the "rules.mk" file. 

`BOOTLOADER=atmel-dfu` may be the right option here, but I'm not sure.

belak, 2017-12-16:
The bootloader shouldn't be overwritten as far as I know. Input:Club uses a custom bootloader.

However, I reverted this a while back which may have broken it. I thought I tested it but maybe there were stale objects or something.

https://github.com/qmk/qmk_firmware/commit/eb98af673a999ede17e293e00220b43d717d500f

belak, 2017-12-18:
@tengg What makes you think the magic jump to bootloader isn't working? As far as I can tell, this port is working on my K-Type. The one oddity is that only the left USB port works.

drashna, 2018-03-25:
Looks like the K-Type is using Input Club info, but the other boards are not. 

I'd add them myself, but I don't own a board to verify.

PR #2669 (2018-04-04): Update input:club boards to use the proper vendor and device
Fixes #2143 (At least all the I:C boards I know of)

The device name info comes from https://github.com/kiibohd/configurator/blob/master/src/kii/device/keyboard.cljs

There were a few oddities... One I saw was with the Infinity 60%... in the configurator, the versions with and without LEDs are separate, but they're not separate here. I set the name to the non-LED version, but we may want to revisit that. Additionally, it looks like all I:C keyboards use the same vendor and device IDs... which is a little odd, but it makes this work e [... cut]
closed 2018-04-04 by jackhumbert