qmk/qmk_firmware #2146: add a startup audio with RGB enabled?

YCF, 2017-12-13:
we can't enable audio and rgb at the same time?
how about just a startup sound,and then disable it?

drashna, 2017-12-13:
You can, and I have. 

The problem is that you risk the firmware being too large (atmega32u4, so pro micro and teensy 2.0 and others), which will soft-brick the controller. 

As long as the firmware is small enough to fit, then you can get both enabled.

YCF, 2017-12-13:
@drashna would you please show me some code,thanks.

drashna, 2017-12-14:
Check out the orthodox in the repo. My keymap stuff is there, actually. 
https://github.com/drashna/qmk_firmware/blob/master/keyboards/orthodox/keymaps/drashna/rules.mk

If you run this: `make orthodox/rev1:drashna RGBLIGHT_ENABLE=yes AUDIO_ENABLE=yes FAUXCLICKY_ENABLE=no`, it will be RIGHT under the limit by about 200 bytes. Very, very close. 

That said, I also use `EXTRAFLAGS=-flto`, which is apparently some code optimization stuff that shaves a bit of space, but can break your keymap (it doesn't like `fn_actions`, for instance)

But basically, just start turning stuff off until it's small enough to fit. And make sure that you've added the `#define C6_AUDIO` or `#define B5_AUDIO`, depending on your board.

YCF, 2017-12-14:
@drashna 
ps: I'm working on my planck-handwired,powered by pro micro.
I will check it out.
orthodox is awesome！
Thanks a lot.

closed 2017-12-14 by YCF

drashna, 2017-12-14:
Yeah, turn off stuff in the `rules.mk` that you don't need, until it fits. 

Also, GB for the Orthodox: https://www.reddit.com/r/MechanicalKeyboards/comments/7i2ler/10_days_left_for_the_orthodox_group_buy/