qmk/qmk_firmware #2152: Make has a strange shell "expecting primary" error

cbbrowne, 2017-12-14:
This seems to be a cosmetic problem, but it's annoying.

`cbbrowne@cbbrowne2 ~/G/qmk_firmware> make xd75:cbbrowne
QMK Firmware gh60-0.1
Making xd75 with keymap cbbrowne

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   24364       0   24364    5f2c ./.build/xd75_cbbrowne.hex

Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/xd75_cbbrowne.elf                                                                   [OK]
Creating load file for flashing: .build/xd75_cbbrowne.hex                                           [OK]
Copying xd75_cbbrowne.hex to qmk_firmware folder                                                    [OK]
sh: 1: arithmetic expression: expecting primary: ""
sh: 1: [: -gt: unexpected operator
cbbrowne@cbbrowne2 ~/G/qmk_firmware>`

Searching for the error message suggests this would happen if you try to evaluate shell expressions in dash when you assumed you were running Bash.

It's not about building my keymap; if I just run "make" the error messages pop up for each keyboard that is built.

My best guess is that this is about checking firmware sizes, in the "check-size" target; the code in tmk_core/rules.mk has a -gt operator.

I get nothing other than the error message if I ask for check-size:

`cbbrowne@cbbrowne2 ~/G/qmk_firmware> make planck:default:check-size
QMK Firmware gh60-0.1
Making planck/rev5 with keymap default and target check-size

sh: 1: arithmetic expression: expecting primary: ""
sh: 1: [: -gt: unexpected operator
cbbrowne@cbbrowne2 ~/G/qmk_firmware>`

skullydazed, 2017-12-22:
I think #2189 fixes this. Can you give it a try?

legrostdg, 2018-02-03:
It fixes the issue for me.

PR #2352 (2018-02-08): make check_size work across platforms
fixes #2252
fixes #2152 
fixes #2307 
closes #2189 

There have been some troubles making check_size work across all platforms used to compile QMK.

check_size is tricky because it needs to discover how much space is available on the chip being flashed, to calculate if the hex file it just compiled is too big. That data is set into the `FLASHEND` C macro at compile time by the C preprocessor, but only when invoked with definitions that avr-gcc sets in response to the -m flag. So we can't simply look it up from the environment, nor can we compile a host-nat [... cut]
closed 2018-02-08 by jackhumbert