qmk/qmk_firmware #2163: avrdude: not found

mattrberry, 2017-12-17:
I am trying to flash my Iris keyboard using the command `make iris/rev2:default:avrdude` as described in the readme. After resetting the controller, I get the message:

```
QMK Firmware 0.5.195
Making iris/rev2 with keymap default and target avrdude

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  25490	      0	  25490	   6392	./.build/iris_rev2_default.hex

Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/iris_rev2_default.elf                                                               [OK]
Creating load file for flashing: .build/iris_rev2_default.hex                                       [OK]
sh: 1: arithmetic expression: expecting primary: ""
sh: 1: [: -gt: unexpected operator
Detecting USB port, reset your controller now.....
Detected controller on USB port at /dev/ttyACM0
sh: 15: avrdude: not found
tmk_core/avr.mk:172: recipe for target 'avrdude' failed
make[1]: *** [avrdude] Error 127
Make finished with errors
Makefile:528: recipe for target 'iris/rev2:default:avrdude' failed
make: *** [iris/rev2:default:avrdude] Error 1
```

Am I missing a library or something? Thanks for any help!

drashna, 2017-12-17:
Yeah, this is a known issue.  You can manually install/download it, and it will pick it up. 

Also, this is on a Mac?  If so, part of the failure is due to the way that grep is being used.

mattrberry, 2017-12-17:
I haven't worked with Arduinos before. Would you mind pointing me to some documentation to manually install it?

And nope, this is on Ubuntu!

drashna, 2017-12-17:
Edited, are you using dash or an alt to bash? As that seems to cause issues. 

As for getting avrdude, do this: http://ubuntuhandbook.org/index.php/2014/09/install-avrdude-6-1-ubuntu-1404/

Then run "avrdude" to see if it's working (I believe).  If it is, rerun the make command above, and it should "just flash it", once the board is in bootloader mode.

mattrberry, 2017-12-17:
I generally use fish and thought that might have been the issue, but I get the same error with bash.

I'll give that a shot and see if it works! Thanks for the help!

mattrberry, 2017-12-17:
That was it! Thank you!

closed 2017-12-17 by mattrberry

VerstandInvictus, 2018-04-20:
For anyone in the future who comes across this and is using MSYS2, the incantation you want is

`pacman -S mingw-w64-x86_64-avrdude`

Cway14, 2021-07-21:
The solution for MacOS was not posted but is very simple:

```brew install avrdude```