qmk/qmk_firmware #2164: How to set NUM_LOCK On regardless it's state?

K-rnivoro, 2017-12-18:
For my Commodore 64 (and it's joysticks) project, I need to set Num Lock to be always on in order to get the joysticks mapped as numpad keys. Thing is I don't want/need the Num Lock key in my keymap because there is no numpad, so I need a way to check the Num Lock status and activate it in case it is off.\
Any ideas?

drashna, 2017-12-18:
This is pretty simple to do, and I do so for my numpad:
https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/woodpad/keymaps/drashna/keymap.c#L111

Just add this to your keymap file:
```
void led_set_keymap(uint8_t usb_led) {
  if (!(usb_led & (1<<USB_LED_NUM_LOCK))) {
    register_code(KC_NUMLOCK);
    unregister_code(KC_NUMLOCK);
  }
}
```

K-rnivoro, 2017-12-19:
Good! I'll try it ASAP!!
Thanks Drashna!

drashna, 2017-12-19:
Welcome! 

And just to explain what's happening, the "led_set_keymap" is called in initialization and any time the "lock" state is changed for any of the states.  

This checks the NUM LOCK state, and if it's disabled, sends the "numlock" key press to enable it.

skullydazed, 2018-02-01:
Did this work for you @K-rnivoro?

K-rnivoro, 2018-02-01:
Oh, sorry I didn't reply!! It worked like a charm!! Thanks Skully!!

lukexorz, 2018-03-08:
If I wanted to just check numlock status one time when my board is plugged in and set numlock on if it is off, how would I do that? I'm using an Ergodox EZ if that makes any difference

Also, drashna, I keep running into you for help with QMK (here and on reddit), just want to say I appreciate your help a lot!

drashna, 2018-03-08:
Untested: 

```
void matrix_init_user (void) {
  if (!(host_keyboard_leds() & (1<<USB_LED_NUM_LOCK))) {
      register_code(KC_NUMLOCK);
      unregister_code(KC_NUMLOCK);
  }
},
```

I think that should force numlock on, but only when the keyboard starts.

K-rnivoro, 2018-03-08:
Yes. I tried and works ok ONLY when the keyboard starts.
Try with two KB. Turn off NumLock with a normal one and plug the hacked one
after that.

drashna, 2018-03-25:
Any issues with having this closed?

K-rnivoro, 2018-03-25:
No. Sorry. Do I close it or you do it?

drashna, 2018-03-25:
Either of us can.  I've just been cleaning up issues.

I've closed it now. If needed, reopen it or open a new issue.

closed 2018-03-25 by drashna

K-rnivoro, 2018-03-25:
I'm fine with the solución you gave me.
Anyway I found that VICE emulator works either way, doesn't matter if
NUMLOCK is on or off.
I've had a very warm reception on the Retro Computer Community with my
Commodore 64 keyboard. (Turns out to be better than a commercial stuff like
Keyrah).
I'm planning to publish it, also as part of qmk, would you help me with the
formalities and stuff?

Thanks for everything.

drashna, 2018-03-25:
Awesome. 

And nice to hear that. 

As for making part of QMK, definitely.  It should probably go into the handwired section though. 
And if you want/would, create a new issue for discussion on that.

u17194907425, 2021-08-12:
I hope it is OK to use this issue to continue answering the initial question as the initial one seems to get outdated.

This code works now. The code posted by @drashna 3 years ago does not seems to work any more

```c
void led_set_user(uint8_t usb_led) {
  if (!(usb_led & (1<<USB_LED_NUM_LOCK))) {
    tap_code(KC_NLCK);
  }
}
```

Note that the `led_set_user` seems to be marked as deprecated in here https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_led_indicators

K-rnivoro, 2021-08-12:
Thank

> I hope it is OK to use this issue to continue answering the initial question as the initial one seems to get outdated.
>
> This code works now. The code posted by @drashna 3 years ago does not seems to [...]

Thanks!!! I'll try it!

Shfty, 2021-09-21:
As a heads-up to anyone else looking to force numlock, the code above will render a ZSA Moonlander unusable if added to a custom keymap.c

I'm unsure if this is an artifact of using the ZSA fork for compatibility with their Oryx software, or a consequence of some underlying LED implementation. The board powers on and displays a single blue LED, but is otherwise unresponsive until being reflashed with a backup firmware.

Myridium, 2021-10-04:
Does anyone have a solution that works today??

K-rnivoro, 2021-10-04:
Kind of funny I wrote this question almost four years ago and still no
absolute answer...

fauxpark, 2021-10-04:
The problem is that the num/caps/scroll lock state is not managed by the keyboard, so all you have to go on is whether the host has told the keyboard to turn on the LED, and the exact behaviour is inconsistent across OSes. On Windows, for example, the state is shared across all keyboards, but macOS treats each keyboard individually and doesn't recognise num lock as num lock anyway.

Zereef, 2022-01-13:
So does any have a good way to do it?
NUM_LOCK always on regardless of its state

ErinCall, 2023-04-15:
Hello from the future! I have this working. [Drashna's "untested" suggestion](https://github.com/qmk/qmk_firmware/issues/2164#issuecomment-371586236) was largely correct, except for putting it in `matrix_init_user`. It seems like that's just too early in the process.

I have my numpad on a separate layer, so I moved the "turn it on if it's off" code to `layer_state_set_user`:

```c
uint32_t layer_state_set_user(uint32_t state) {
  switch(biton32(state)) {
  case NUMPAD:
    // turn on numlock, if it isn't already on.
    if (!(host_keyboard_leds() & (1<<USB_LED_NUM_LOCK))) {
      register_code(KC_NUMLOCK);
      unregister_code(KC_NUMLOCK);
    }
    break;
  }
  return state;
};
```

If you aren't using a separate layer I imagine you could do something similar in `process_record_user`. Sending a keypress depending on the LED state is a sound approach, it's just a question of when and where it happens.

(I don't know if any of y'all're still thinking about this, but since this is one of the top search results I figured I'd fix the [DenverCoder9](https://xkcd.com/979/) situation 🙂)

K-rnivoro, 2023-04-16:
Thanks Erin!! It seems this has been around for years. Look how old my
question is...

u17194907425, 2024-02-23:
It looks like now the above methods with `USB_LED_NUM_LOCK` will no longer work.
For me the following works in the `keymap.c`:

```c
bool led_update_user(led_t led_state) {
	if (!led_state.num_lock) {
		tap_code(KC_NUM_LOCK);
	}
	return true;
}
```