qmk/qmk_firmware #2178: In-keyboard layout versus OS input source

veikman, 2017-12-19:
I am coming to QMK from an OS-side remap of QWERTY to Colemak. One of the reasons I use Colemak is its support for e.g. Swedish characters on (right Alt) “layer 3”, without the awkwardness of Unicode hex entry or the programmer-unfriendly standard Swedish layout.

I initially assumed I would be creating a QMK keymap with the Colemak layout built into it, while reverting to a standard QWERTY mapping on the OS side. However, there seem to be tradeoffs to every option on the OS side. For example, US English QWERTY has no layer 3 in GNOME. I’d be (re)creating layers 3 and 4 of Colemak in process_record_user(). Pros and cons of this would be:

+ Pro: QMK Unicode/macro entry works as intended.
- Con: I’d have Unicode pseudo-macro sequences for at least some of e.g. [äåö“”—] in place of the OS-side map. Slow and not portable beyond IBus.
- Con: On a laptop, I would have to switch between layouts in the OS upon docking/undocking to continue using Colemak on the inferior built-in keyboard.

The alternative is to keep using Colemak on the OS side, with QWERTY in the QMK keymap.

+ Pro: The OS can name and show roughly the right layout. Good if you forget some layer 3 stuff and don’t write it all on your keys.
+ Pro: Laptop undocking etc. is smooth.
- Con: Unicode/macro entry in QMK requires awkward redefinitions of unicode_input_start() and hex_to_keycode(), reversing the OS’s remap. The specifics would again be married to the OS layout.

Any non-QWERTY layout will run into this. Is there a preference for either solution in the community? Am I missing something?

If the second alternative (QWERTY on QMK, non-QWERTY in the OS) is considered viable, I propose centralized **support for layout-specific reverse maps** e.g. in process_unicode_common.c, for an enum of popular alternatives to QWERTY. If not, it would be nice if the problem were apparent in the documentation of the Unicode and macro features.

drashna, 2017-12-19:
I'm not  sure if this is helpful or not, but ...

If you haven't seen this: https://github.com/qmk/qmk_firmware/blob/ef2961798cddb43dbcbac9d72ca9d84d71ad565e/quantum/keymap_extras/keymap_colemak.h

And you can use "CM_" instead of "KC_" and it should map to the correct key when using software colemak.

veikman, 2017-12-19:
Thank you drashna, that is indeed helpful! That takes care of macros, but I see no equivalent for Unicode, which was my main interest. I have updated the original description to reflect this.

drashna, 2017-12-19:
Glad I could help with at least part of that. 
The unicode stuff I'm not familiar with, so I can't help there. Sorry. 

But in case you haven't seen it: 
https://docs.qmk.fm/feature_unicode.html

And IIRC, there are a number of custom keymaps that handle this, if you're willing to dig.

drashna, 2017-12-19:
Actually, this may be what you're looking for: 
https://github.com/qmk/qmk_firmware/blob/d2ff66a985b938e87fffe55c1d9f1dc55e356f91/layouts/community/ergodox/familiar/keymap.c#L101

veikman, 2017-12-20:
To clarify: I think I know how Unicode input works in QMK. The problem is that QMK is necessarily limited to USB HID codes and ignorant of what the OS is going to do with them. For example, to get a proper right-side quotation mark, I can use UC(0x201D). For Linux/IBus, by default, QMK will then send a sequence of:

> Ctrl+Shift+u 2 0 1 d

The OS receives this and misreads it as:

> Ctrl+Shift+l 2 0 1 s

To get the right result I anticipate that misreading and compensate for it, using the following overrides in my keymap file.

```
void unicode_input_start (void) {
  register_code(KC_LCTL);
  register_code(KC_LSFT);
  register_code(CM_U);
  unregister_code(CM_U);
  unregister_code(KC_LSFT);
  unregister_code(KC_LCTL);
  wait_ms(UNICODE_TYPE_DELAY);
}

uint16_t hex_to_keycode(uint8_t hex)
{
  if (hex == 0x0) {
    return KC_0;
  } else if (hex < 0xA) {
    return KC_1 + (hex - 0x1);
  } else if (hex == 0xD) {
    return CM_D;
  } else if (hex == 0xE) {
    return CM_E;
  } else if (hex == 0xF) {
    return CM_F;
  } else {
    return KC_A + (hex - 0xA);
  }
}
```

I'm suggesting that overrides like these, but less destructive, could be implemented centrally in QMK in a manner analogous to the present sendstring_colemak.h file. The individual user could then import a future unicode_colemak.h, or set a more general OS layout option in their config.h.

drashna, 2018-03-25:
Yeah, there isn't a good solution for this, other than maybe different layers for each OS... because of the non-standardized implementation.

closed 2018-10-22 by drashna