qmk/qmk_firmware #2191: jj40 send string limit

danielo515, 2017-12-23:
Does anybody know if jj40 has a limitation on the number of characters it can send through send string ?
it is stopping every send at 5 characters length and after that it does not execute the next instruction, which is sending enter key.

I don't know where to look limitations and features of this keyboard.

Thanks in advance

drashna, 2017-12-25:
Try adding more characters before the enter, and see if that works.

danielo515, 2017-12-25:
I'm using the function send string,not the macro. Do you mean use the macro for all?

drashna, 2017-12-25:
Yeah, add more characters to the `SEND_STRING` function.  That or try `send_string`, and see if that helps (note the case, it does matter)

danielo515, 2017-12-27:
Hello @drashna ,
as you can see here i'm trying both things:
https://github.com/danielo515/qmk_firmware/blob/keyboard/jj40/keyboards/jj40/keymaps/split/keymap.c#L174-L210

None of them work. If you have any idea I'll ve very grateful.
Regards

danielo515, 2018-01-03:
Hello again @drashna 
Today something unexpected happened: the macro sent the enter keycode.
It was weird, so I made some test on a text editor. If I execute the macro several times in a row, after 4 executions it sends the enter keycode,  then it works a couple of times (between one and two) and then the same behavior again.
I'm not sure if this has anything to do with the fact that it is triggered from a layer (LT macro) so when the keycodes are sent, the other key is also pressed.

Regards

drashna, 2018-01-03:
Maybe try:
```
SEND_STRING("togit");
_delay_ms(10);
SEND_STRING(SS_TAP(X_ENTER));
```

I do this for a ton of my macros due to odd timing issues with games.  It may help here, until somebody builds in a delay option into `SEND_STRING`.

danielo515, 2018-01-05:
Hello @drashna 
Thank you for your suggestion, I tried it and the result is the same.  I don't understand what's wrong with this keyboard. It has weird behaviors since day 0

danielo515, 2018-01-18:
Hello again.

I didn´t tested this in more detail, but it has started working suddenly.
They only things that have happened were:

- I have flashed the firmware from a Mac computer instead of my usual windows machine
- I added support for underglow following the instructions at #2224

Apart from that, everything is the same. I'm scared about this stopping working randomly on the future. The changes can be seen here:

https://github.com/danielo515/qmk_firmware/compare/keyboard/jj40...danielo515:feature/nonSplit

ryanm101, 2018-01-29:
Not working  for me
```
SEND_STRING ("git checkout");
```
results in 
```
git chec
```

using `https://github.com/krusli/qmk_firmware` as he has JJ40 tapdance working

danielo515, 2018-01-30:
Sorry, I was unclear. What started working was sending the enter key after a short string of less than 5 characters.

ryanm101, 2018-02-13:
Just tried
```
void git_dance (qk_tap_dance_state_t *state, void *user_data) {
  if (state->count == 1) {
      register_code(KC_G); unregister_code(KC_G);
      register_code(KC_I); unregister_code(KC_I);
      register_code(KC_T); unregister_code(KC_T);

      register_code(KC_SPC); unregister_code(KC_SPC);

      register_code(KC_C); unregister_code(KC_C);
      register_code(KC_H); unregister_code(KC_H);
      register_code(KC_E); unregister_code(KC_E);
      register_code(KC_C); unregister_code(KC_C);
      register_code(KC_K); unregister_code(KC_K);
      register_code(KC_O); unregister_code(KC_O);
      register_code(KC_U); unregister_code(KC_U);
      register_code(KC_T); unregister_code(KC_T);

      //register_code(KC_ENT); unregister_code(KC_ENT);
    }
}
```
still results in 
`git chec`

drashna, 2018-02-24:
The issue is here: https://github.com/qmk/qmk_firmware/blob/a258358b6c2df0d1347a6506831076ef6a9784b3/tmk_core/protocol/vusb/vusb.c#L49

It's virtual USB, and it's dropping stuff rather than waiting.

skullydazed, 2018-03-01:
Capturing more context about this issue. From the gitter chat @fredizzimo shared this information:

> It should loop for a while, waiting for the USB to become available
> it could be as simple as changing the if to a while loop
> but I fear that we at least need to add some break out of it if the USB is disconnected
> and there's also probably some poll function that needs to be called

We need someone who has an atmega32a board to write and test this code, or an atmega32a board needs to make its way to one of the qmk developers in order to properly fix this problem.

danielo515, 2018-03-01:
I would love to test that code myself. Sadly my atmega32a board has it's boot section overwritten an I am unable to flash it.

drashna, 2018-03-01:
Oh no!   If the board has ISP/ICSP header, you may be able to ISP flash the bootloader (if you can find a hex for it)

ccristob, 2018-03-31:
Just encountered this issue with a YMD96 as well, hoping a board makes its way to QMK devs as well.

rasmusx, 2018-04-06:
Hey everybody!

I have created are MR that should fix this issue, but it would be good to get some more feedback and testing if possible.

Patch is tested with Leeku L3 pcb that runs on ATmega32A and seems to be working fine without any side effects.

ccristob, 2018-04-06:
Thanks! Just tried it out on a a YMD96 and send_string works with the full string now. Will try it out for the next couple of days to see if I encounter any issues. Thanks again.

rasmusx, 2018-04-07:
@ccristob nice to hear that it works out for you also!
Please let know if you encounter any issues.

christopherjanzen, 2018-04-30:
@rasmusx I'm having similar issues with my Pearl and unexpected tap dance behaviour. I've tried your fix but it isn't helping. The issue I'm experiencing is that on a single tap it should print a left bracket and on double tap it should print a left parenthesis. As of right now what happens is on double tap the left parenthesis is printed properly but when doing a single tap the left bracket is not printed until another key is pressed. Any thoughts on this?

PR #2697 (2018-07-22): VUSB fix report dropping if usbInterruptIsReady() returns false
This commit should fix #2191.

I have tested it out with Leeku L3 PCB that runs with ATmega32A. So far I haven't seen any side issues caused by this drop and macros with SEND_STRING are working now, but it would be nice if someone else could also test it out or give some feedback.

Marked MR currently as WIP before getting some more feedback.
closed 2018-07-22 by jackhumbert