qmk/qmk_firmware #2194: Backlighting Effects

jeffmhastings, 2017-12-26:
Does QMK have support for different backlighting modes? EasyAVR has static, breathing, reactive, and erosion. I've looked but haven't really seen anything other than toggles and levels.

Duckle29, 2017-12-28:
Check the animations table here: https://docs.qmk.fm/feature_rgblight.html#rgblight-keycodes
You can also make your own

drashna, 2017-12-28:
@Duckle29 That's RGB underglow, not backlighting.  

And no, I don't think it has multiple modes, other than "on/off/breathing"

Duckle29, 2017-12-28:
Ah okay. My mistake. I was under the impression the RGB stuff was all unified, but I can see I was wrong

Poluglottos, 2018-01-10:
I believe all that's currently available is 'breathing' and then static brightness levels, 0 through 9

skullydazed, 2018-02-01:
We don't currently have support for those modes, mostly because no one has written them. If someone would like to take on that task this can serve as a placeholder for that work.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc