qmk/qmk_firmware #2195: Dynamic re-assignment of collums and rows

Duckle29, 2017-12-27:
Hello there, I'm working on a new split keyboard, based on an idea sparked by someone I talked to on IRC.

It uses only one microprocessor and instead relies on the interconnect to carry the matrix over, to allow the master half to scan it. Connector choices were discussed, and anything from DB25 connectors to mini-DisplayPort was discussed, but due to size and price, USB-C was chosen as the goal.

Now, this obviously presents a problem. USB-C is reversible. This is very nice for end users, but it causes routing issues, none that can't be overcome though.

It just requires thoughtful pin-layout on the USB-C connector and a way to sense rotation.

The rotation will be sensed like so:
[image]

This method allows the controller to detect that the connector has been rotated, and will allow for the controller to compensate for that.

It is shown with a 32k ohm pullup, as this is the equivalent of the internal pull-ups in the AVRs, the 500R resistor to GND is there to protect the chip if for some reason the pin connected is set to a high output.

A keyboard with this layout would require that QMK supports detecting this change, and changing the pin assignments for columns and rows.

A bit of C to describe the way I'm imagining it could work:
```C
#define ROW_NUM 5;
const uint8_t multi_rows[2][ROW_NUM] = {{C7, D4, D6, D7, F0},{F0, D7, D5, D4, C7}};
const uint8_t rows[ROW_NUM];
// ISR For detecting edges on isFlipped pin
void flipped_isr()
{
    bool is_flipped = readPin(is_flipped_pin);
    for(int i=0; i<ROW_NUM; i++)
    {
        rows[i] = multi_rows[is_flipped][i];
    }
}
```

An actual example of such a layout is shown here: https://imgur.com/a/SFKPi 
Ignore that I'm using a diode in there, the design has been changed to the example in the first image.

As you can see, care has been taken to ensure that cols are cols, and rows are rows, regardless of rotation, with the exception of the spare leftover pin called extra. Personally, I see this as a possible pin to be used for RGB data signal across halves.

Preferably the implementation in the firmware shouldn't need this care with the layout though, and if implemented as shown in the example code above, wouldn't need to. 

I'm am opening this issue to ask for help on how to best implement this feature.

I want to add a note to a particular page of the linked schematic 

[image]

As you can see the cols and rows from the slave halves are named in the following format:
    [function] / [rotated function]

An example is 
    col0 / col7

The naming means that that pin is col0 if it's not rotated, and col7 if it is.

Duckle29, 2017-12-27:
I should also note that the implementation should somehow allow keyboard implementers to reassign special pins in a function, if they need to, an example being the RGB data pin.

Taudris, 2018-01-25:
It seems like a really bad idea to use USB C in this way. What happens if someone decides to plug an actual USB C device into your keyboard? Say, a laptop, or for some reason, a charger? You need to make sure this kind of attempted usage won't result in damage to any of the involved devices.

Aside from that... One thing you could do is have one key in your matrix be always pressed. This solves both connection detection and orientation sensing. Since each key is in series with a diode, it will only work in one direction, so in your matrix scanning, you could first test that one key before continuing the scan. This also avoids the complexity of watching a pin with an interrupt (a pin that could be quite noisy during plug/unplug).

If your matrix is full, you could put a diode in series with a resistor between symmetrical pins. This will require two GPIOs, but allows you to know both whether the slave is connected and its orientation.

Duckle29, 2018-01-25:
@Taudris All very good points. I'll certainly have to consider all of them when I get back to this project. I have the power pins as power pins, to avoid issues if plugged into a computer. I'll have to make sure the firmware doesn't put voltages on any other pins unless it's sure that it's hooked up to a slave. 

I still think it's a good idea, as it's a cheap cable, and it's compact, though it definitely has to be designed with care.

zvecr, 2022-08-28:
Closing due to age

closed 2022-08-28 by zvecr