qmk/qmk_firmware #2197: M65-A Key Swaps: Unwanted Bootmagic Activation

kwkroll32, 2017-12-28:
There have been a few of us with the native QMK M65-A who have had bootmagic unintentionally activated. Symptoms include swapped ESC and GRV, swapped Backspace with Backslash, and other bootmagic features described [here](https://docs.qmk.fm/feature_bootmagic.html). It only ever happens to me on a fresh boot - never during normal use of they board. 

There are a couple of workarounds that return normal key functionality: 

1. reset eeprom and reflash as described in [quantum/tools/](https://github.com/qmk/qmk_firmware/tree/master/quantum/tools).

2. assign "MAGIC_UNSWAP_*" keys to the keymap and turn off each individual swap when they become active.

3. flash the jc65 firmware with bootmagic turned on [i.e. "BOOTMAGIC_ENABLE=yes" in rules.mk]. When the swapped keys get activated, unplug and re-plug the keyboard in with space + backspace pressed. 

These workarounds will relieve symptoms, but it would be ideal to eliminate the root of the problem. Is it possible to truly disable bootmagic, since the bootmagic toggles can apparently still be activated when "BOOTMAGIC_ENABLE" is set to "no"? 

@jetpacktuxedo did you put the jc65 QMK code together? Any clue what is going on here?

drashna, 2017-12-28:
The problem is (IIRC), this can STILL happen.
The "COMMAND" functionality is what is probably causing this.  

If you add `COMMAND_ENABLE = no` to your rules.mk, or add `#undef IS_COMMAND` to your config.h file, it should prevent this from happening

jetpacktuxedo, 2017-12-28:
For the two things that you specifically named, esc->grave and backspace->backslash, those might possibly be explained by the default keymap. By default, esc is KC_GESC which is escape normally but tilde when shifted, for example. though since you say it only happens on a fresh boot I would guess it probably isn't that.

I haven't hit this myself and am generally pretty unfamiliar with the bootmagic functionality. Hopefully someone else knows more.

jetpacktuxedo, 2017-12-28:
Can you try wiping the eeprom next time you reflash it? The process is described here: https://github.com/qmk/qmk_firmware/tree/master/quantum/tools.

That should let us see if it is just something funky that gets left in the eeprom when it ships from the factory.

kwkroll32, 2017-12-29:
@drashna I've set COMMAND_ENABLE to "no" as you suggested and flashed again. I'll have to wait and see if that helped, as I am not able to reproduce the issue on demand. 

@jetpacktuxedo no worries, I was just wondering whether you knew something about bootmagic that I didn't. I have tried wiping the eeprom [workaround # 1 above]. That restores normal behavior, but does not stop bootmagic toggles from being randomly activated again in the future. 

The key swapping only happens to me once every 2-3 weeks. I'll leave this open for a few more days to allow for any other suggestions and see if the issue reappears.

internetirl, 2018-05-20:
I see this was closed.. was it ever resolved or did you ever encounter the issue again? I'm experiencing similar problems.

kwkroll32, 2018-05-20:
You're using the QMK native board, right? Mind linking to your keymap? I still have this problem every few weeks but am not sure whether it is a software or hardware problem. Seems like bootmagic toggles are being set somehow even though they're not on my keymap. I am working around the issue with this reset procedure:

1. unplug keyboard
2. hold space and backspace *** (see note)
3. plug in keyboard
4. repeat as necessary

It usually takes 2-3 tries before I am returned to normal functionality. 

***note that the key I hold for backspace has to be the one that currently appears mapped to backspace. That is, if the "swap backspace and backslash" bootmagic option appears on, I have to press backslash in order to send the backspace keycode.

internetirl, 2018-05-20:
Yup, using QMK.

I made a very basic keymap when I first got the board. I'm not doing anything crazy.
https://github.com/internetirl/qmk_firmware/tree/master/keyboards/jc65/v32u4/keymaps/deeza

Thanks for the workaround. I'd like to also identify the root cause. I'll give that a try if/when I encounter this again.

Do you sleep your computer? I rarely shutdown/restart. I regularly switch keyboards while my PC is on (Windows 10), but I don't hold any keys down while doing so. My workflow usually is simple: wake PC from sleep, leave on most of day, sleep before bed. The one thing I did notice that I'm guessing has nothing to do with it -- the night before this issue happened, I left the RGB lights on. For some reason, if you sleep your computer, it doesn't turn the lights off on the board, so I have a toggle switch in my keymap (fn + q) to turn them off. That was the only night I left them on, but I don't think it's related.

kwkroll32, 2018-05-20:
You might need to add `BOOTMAGIC_ENABLE=yes` to your rules.mk in order for the reset procedure to work. Space + backspace is actually a [bootmagic function](https://docs.qmk.fm/features/bootmagic). 

I rarely sleep my PC, but usually shutdown at the end of the day. I never unplug the keyboard except for that reset procedure. My keymap is pretty busy with layers, functions, and features like tap dance. I always thought this behavior could be an edge case bug in one of those, but your map looks entirely straightforward. 

I am more suspicious that this is a hardware problem now. It doesn't seem to be related to keymap or how you use the board.

jetpacktuxedo, 2018-05-20:
For what it's worth, I think the issue is actually related to the underglow. I have seen similar issues with corrupted eeprom on all of my boards with underglow (though typically it is just underglow stuck in an odd state rather than substituting bootmagic stuff). I asked if anyone who knows about eeprom in the qmk gitter had any ideas, but no one's gotten back to me yet.

kwkroll32, 2018-05-20:
That makes sense to me. I noticed that the problem persists if I just reflash the firmware, and that I have to fully reset eeprom to get it back to normal. Is there an open incident for the underglow eeprom corruption? I'll try disabling underglow to see if that helps.

jetpacktuxedo, 2018-05-20:
Not as far as I know, it's just what I've begun to suspect. I know RGB underglow does a lot of eeprom io, which is why it is my leading guess. if you test with your backlight off and notice a change in frequency definitely update here, because that would basically confirm my theory.

internetirl, 2018-05-20:
Yeah, I had a weird feeling about leaving my lights on overnight while slept. Maybe I'll try to repro by sleeping/waking with lights on/off.

jameshly, 2018-05-29:
Hey, thought I'd add my experience with this issue. 

The bootmagic issues usually only happen to me when I unplug the keyboard from my computer and plug it back in, usually after rebooting. This usually happens when I do something like relocating my computer or plugging the keyboard into a different computer.

I usually remedy this by running the following commands:
 `sudo dfu-programmer atmega32u4 erase`
`sudo dfu-programmer atmega32u4 flash-eeprom quantum/tools/eeprom_reset.hex` 
`sudo make jc65:<insert keymap here>:dfu`

Side note - if you're still getting bootmagic issues _without _  unplugging your keyboard, does your computer provide power to the board when it's asleep/shutdown? Mine does, so that might be why my issue is popping up more "reliably". That might have something to do with it.

internetirl, 2018-06-03:
> Side note - if you're still getting bootmagic issues _without _ unplugging your keyboard, does your computer provide power to the board when it's asleep/shutdown? Mine does, so that might be why my issue is popping up more "reliably". That might have something to do with it.

Yes, this. Exactly. I mentioned earlier in the thread that I have to usually turn off the LEDs before sleeping otherwise they'll just stay on all night. It happened to me again recently and I believe it's when I left the lights on. I haven't had time to run through repro scenarios, but I'll try soon.

kwkroll32, 2018-06-23:
It's been about a month since I disabled backlight and RGB, and I haven't had this issue since then. It's hard to say that the problem is definitely gone, but I'm reasonably sure at this point. 

What are the next steps for resolving this issue? Should I do any other kind of testing? I am satisfied with this workaround for now... although I do have the nylon base and it would be nice to have underglow.

jetpacktuxedo, 2018-06-23:
So for me, that basically confirms that this is related to the EEPROM writes that underglow does. I'll talk to some people and do some research and see if anyone has any ideas for a real solution.

drashna, 2018-06-23:
RGB Underglow has been overhauled, and there are a bunch of `noeeprom` functions that may help. 
https://docs.qmk.fm/#/feature_rgblight?id=led-control

Additionally, there is the `RGBLIGHT_SLEEP` feature too. 
https://docs.qmk.fm/#/feature_rgblight?id=optional-configuration

Additionally, it looks like this board may use the atmega32a MCU. Are you using the 32a version?  If so, there is a bug in the VUSB support that may be causing odd issues, as well. Applying the changes from this PR may help: 
https://github.com/qmk/qmk_firmware/pull/2697

jetpacktuxedo, 2018-06-24:
The 32A PCB is the PS2AVRGB version of this board. This issue is for the QMK revision of the PCB which uses the 32u4. 

My suspicion is that most of the people in this thread are using RGB animations rather than static colors (can anyone confirm?) Which would explain why I almost never have this issue (I just set a static color). Further, I suspect that as some of the color changing animations loop, every cycle when they adjusted hue they write their state to the EEPROM. If the board loses power during an EEPROM write I imagine it could leave the EEPROM in some sort of corrupted state, which could leave some random noise in the EEPROM that gets read as bootmagic state information.

I don't know enough about qmk RGB behavior or EEPROM writes to know whether my guess is even plausible, though. Do you have any ideas there @drashna?

jetpacktuxedo, 2018-10-01:
Hey @kwkroll32 , @internetirl , and @jameshly , can you try recompiling and refashing with the most recent version of qmk (to pull in the changes that @drashna mentioned earlier) and see if it is still an issue? I never managed to reliably reproduce this, otherwise I'd retest it myself.

kwkroll32, 2018-10-09:
I updated my fork tonight and turned backlight back on so we'll see what happens. I had always used a solid color, so I'll keep doing that for testing. 

Also note that I didn't have one instance of this issue for months after I disabled backlight/RGB in commit 46bc2e4 above.

kwkroll32, 2018-10-31:
so far, so good with my recent firmware update. I've had RGB on constantly for >3 weeks of normal usage. I think I'm good if you guys want to call this resolved for now.

jetpacktuxedo, 2018-10-31:
🙏 @drashna, I'm gonna call this fixed I think. Someone can reopen if they can repro it again with a newer qmk version.

drashna, 2018-10-31:
Awesome, glad to hear it! 

And yeah, will close it now.

closed 2018-10-31 by drashna

pyrho, 2019-05-13:
I've had this issue on and off recently, it's not a case of unwillingly pressing the bootmagic keys.
What fixed it for me was @jameshly's suggestion to reflash the eeprom.