qmk/qmk_firmware #2208: Let's split: Clarify/Reference why audio and RGB underglow can't coexist

rickard-von-essen, 2018-01-01:
In the [rules.mk](https://github.com/qmk/qmk_firmware/blob/master/keyboards/lets_split/rules.mk#L67) he comment states:

>  Enable WS2812 RGB underlight.  Do not enable this with audio at the same time.

But there is no explanation why or any reference with more information. It would be great to add something to clarify this.

drashna, 2018-01-01:
I think that there used to be a technical reason for this.

But now, I think it's mostly size.  If the firmware is larger than 28k, it can/will soft brick the controller.  And since both the RGB underglow code and the audio code are fairly large, it's a tight fit, at best.

Taudris, 2018-01-18:
The technical reason appears to have been twofold:

1. `rgblight.c` has some vestigial references to TIMER3 in some debug prints (quoted below), suggesting RGB used to require TIMER3 to function. This file now has a function called `rgblight_task()` which is called in QMK's main loop, and uses `timer_read()` for timing instead of a hardware timer.
2. Audio can use pin `B5` or `C6`. Using `B5` requires TIMER1, and `C6` requires TIMER3. Let's Split uses `B5` in the key matrix, so you're forced to use `C6` and TIMER3.

```c
void rgblight_timer_enable(void) {
  rgblight_timer_enabled = true;
  dprintf("TIMER3 enabled.\n");
}
void rgblight_timer_disable(void) {
  rgblight_timer_enabled = false;
  dprintf("TIMER3 disabled.\n");
}
void rgblight_timer_toggle(void) {
  rgblight_timer_enabled ^= rgblight_timer_enabled;
  dprintf("TIMER3 toggled.\n");
}
```

drashna, 2018-03-25:
Checked the code, the timer code was striped and added as a "task" instead.  

There is no practical reason you cannot use both, other than firmware size.

https://github.com/drashna/qmk_firmware/blob/drashna_keymaps/quantum/rgblight.c#L445-L462
https://github.com/drashna/qmk_firmware/blob/drashna_keymaps/quantum/rgblight.c#L482-L505

drashna, 2018-03-25:
@skullydazed Short of editing ALL of the readme's and rules.mk that include this .... I'm not sure anything needs to be changed...   Cleaning up this may be a good idea, but it will be tedious or wee need a script to do so.

PR #4199 (2018-10-22): Remove outdate RGB - Audio infortmation from QMK
This removes the `# Enable WS2812 RGB underlight.  Do not enable this with audio at the same time.` line from all of the rules.mk files, since this is not longer accurate. 

Closes #2208
closed 2018-10-22 by jackhumbert