qmk/qmk_firmware #2212: Build for Teensy++ 2.0

jawb, 2018-01-02:
I made a new keyboard project and wanted to build the firmware following this guide https://docs.qmk.fm/getting_started_make_guide.html I used **make keyboard:default:teensy** option but the hex I got wasn't accepted on the Teensy loader since it's a build for Teensy 2.0 (ATMEGA32U4) and I have a Teensy++ 2.0 (AT90USB1286).

What option should I use ?

drashna, 2018-01-02:
I'm not 100% sure about this, but ... in the rules.mk for the keyboard, make sure you have this:
```
MCU = at90usb1286
BOOTLOADER = halfkay
```

And see if that helps/works.

jawb, 2018-01-02:
@drashna Thanks that worked, it's my first time starting a a full keyboard project in QMK, shouldn't there be a way to make the keyboard easy to build for any architecture without altering files ?

drashna, 2018-01-03:
I'm glad to hear it!

I'm not knowledgeable enough to give you a good answer for that.

But there are a number of differences between the different platforms.  Especially AVR vs ARM.  

Though, I think part of this was a lack of bootloader size being specified.  But I'm not 100% sure.

jawb, 2018-01-03:
What made it work for me is the MCU = at90usb1286 part, I didn't change the BOOTLOADER because I actually did that before you posted.

My point is the keyboard files should stay the same, and anyone could build without having to alter anything in order to build. ATMEGA32U4, AT90USB1286 or other should be set in make command rather than a file I think.

closed 2018-01-03 by jawb

skullydazed, 2018-01-03:
We consider the MCU to be one of the things that defines a particular keyboard. For an example, compare `keyboards/planck/rev5` to `keyboards/planck/light`. You can do something similar if you want your keyboard to support multiple processors, but selecting the MCU via the make command line is not something we want to explicitly support.