qmk/qmk_firmware #2220: Requires GNU grep

rickard-von-essen, 2018-01-03:
Building on macOS one command fails (but the build succeeds) with:

```
$ make lets_split/rev2:default
QMK Firmware 0.5.206
Making lets_split/rev2 with keymap default
[...]
usage: grep [-abcDEFGHhIiJLlmnOoqRSsUVvwxZ] [-A num] [-B num] [-C[num]]
        [-e pattern] [-f file] [--binary-files=value] [--color=when]
        [--context[=num]] [--directories=action] [--label] [--line-buffered]
        [--null] [pattern] [file ...]
```

This is because macOS ships with BSD grep. Replacing all occurrences of `grep` with `ggrep` (after installing it with `brew install grep`) (`git grep -l grep | grep -v keyboards/ | xargs -I= sed -i'' -e 's/grep/ggrep/g' =`) and the build goes through without the error.

This can ge fixed in either two ways A) require GNU grep and invoking the correct binary (`ggrep` on macOS and `grep` on others) or B) change the specific invocation of `grep` which use an unsupported flag in BSD grep. 

I can do a PR if I get some advice on which solution to pick.

I'll try to find which of the 17 `grep` in the codes uses the incompatible flag and add it as a comment.

drashna, 2018-01-04:
You may want to check out #2189, as it looks like this is actively being addressed

rickard-von-essen, 2018-01-04:
Duplicates #2122

closed 2018-01-04 by rickard-von-essen