qmk/qmk_firmware #2224: JJ40 Underglow & Backlight

ryanm101, 2018-01-04:
When configured the underglow it starts in state off and toggling the HUE will turn it on. 

Backlight LEDs underswitches seem to work when flashing firmware or just after flashing but once unplugged they are off and cannot be toggled with the toggle keys

krusli, 2018-01-08:
Did you try using RGB_MOD to toggle through the modes, and RGB_VAI and RGB_VAD to increase/decrease the brightness (value)? My JJ40 initially didn't seem to fully work with the underglow but either a few flashes or playing around with the keycodes helped and it now works perfectly.

There seems to be some kind of weirdness too I think with the PS2AVRGB boards occasionally. My JJ40 developed a weird issue where pressing the V key would also trigger the Lower key, but a flash using Bootmapper and another back to QMK gets rid of the issue.

Mine is a v1.5 PCB.

ryanm101, 2018-01-10:
I've the 1.5 pcb as well.

I'll try the other keys you mention as well, though i can live with using the key to toggle is the backlight for the LEDs I find more annoying I cant get to work.

krusli, 2018-01-10:
Dug a bit deeper: that part (to send another command setting all 3 color channels to 0) is necessary to toggle the RGB LEDs off with the JJ40. I've traced the code back to `rgblight.c`, and it seems that the current state of the LED underglow lighting is committed to EEPROM via `eeconfig_update_rgblight()`. What's interesting is even though the toggle state isn't saved, the color and mode of the LEDs do seem to stick even after a unplug/replug, so it's not like the board can't write or read anything from EEPROM.

As for the backlight I think it just comes down to figuring out what to send over to the keyboard via i2c to update the level of the backlight LEDs. I don't have backlight LEDs installed on my JJ40 but I'll try to go looking to see if someone's implemented backlighting on another PS2AVRGB board (the code should port over pretty easily).

krusli, 2018-01-10:
I'm currently having a look at #959 (issue by the user who originally ported the PS2AVRGB boards to QMK) and the [PS2AVRGB](https://github.com/showjean/ps2avrU) repo to see if I could port over the backlighting code. Unfortunately I can't test it out myself but I could report back once I get to a state where my changes would have a good shot at working. Once I do that though could you help me with testing the code?

krusli, 2018-01-10:
Could you try changing the backlight pin to D4?

From the PS2AVRGB repo:
```
// PB0-PB7 : col1 .. col8
// PA0-PA7 : row1 .. row8
// PC7-PC0 : row9 .. row16

// PD0 : LED NUM
// PD1 : LED CAPS
// PD2 : D+ / Clock
// PD3 : D- / Data
// PD4 : FULL LED
// PD5 : 3.6V switch TR
// PD6 : LED SCROLL
// PD7 : row17
```

krusli, 2018-01-10:
For the issue with the underglow LEDs at least, go to `jj40.c`, clear it out and paste this in:

```C
#include "jj40.h"

#include <avr/pgmspace.h>

#include "action_layer.h"
#include "quantum.h"

#include "i2c.h"

// custom RGB driver
extern rgblight_config_t rgblight_config;
void rgblight_set(void) {
  if (!rgblight_config.enable) {
    for (uint8_t i=0; i<RGBLED_NUM; i++) {
      led[i].r = 0;
      led[i].g = 0;
      led[i].b = 0;
    }
  }

  i2c_init();
  i2c_send(0xb0, (uint8_t*)led, 3 * RGBLED_NUM);
}

bool rgb_init = false;

__attribute__ ((weak))
void matrix_scan_user(void) {
    // if LEDs were previously on before poweroff, turn them back on
    if (rgb_init == false && rgblight_config.enable) {
      i2c_init();
      i2c_send(0xb0, (uint8_t*)led, 3 * RGBLED_NUM);
      rgb_init = true;
    }

    rgblight_task();
    /* Nothing else for now. */
}
```
The keyboard does remember the last state of the LEDs, however to turn the LEDs on a signal has to be sent via I2C to turn on the underglow on the board.

ryanm101, 2018-01-10:
I'll give it a go when i get home tonight. Thanks for looking into this.

ryanm101, 2018-01-10:
Yes i noticed it saved colour but not state which i thought was interesting. Happy as long as i can config the "default" state in code..

ryanm101, 2018-01-10:
@krusli Ok so underglow is now working on plugin... Good job thanks

I'll test backlight next

ryanm101, 2018-01-10:
Nope pin D4 is doing nothing:(   Using   BL_TOGG, BL_ON, BL_OFF to test

krusli, 2018-01-11:
Could you try setting it to Pin B5 instead for now? In QMK (`quantum.c`) it seems to use the output compare unit `OCR1A` and `COM1A1` as it is done in PS2AVRGB (`setPWM()` calls `timer1PWMBSet()` which sets the value of `OCR1A`.

Actually on second thought I'm rather confident that this has a pretty decent chance of working. QMK already has the low-level AVR PWM stuff implemented so I think that should do the trick.

(for reference, code from `quantum.c`)
```C
// depending on the pin, we use a different output compare unit
#if BACKLIGHT_PIN == B7
#  define COM1x1 COM1C1
#  define OCR1x  OCR1C
#elif BACKLIGHT_PIN == B6
#  define COM1x1 COM1B1
#  define OCR1x  OCR1B
#elif BACKLIGHT_PIN == B5
#  define COM1x1 COM1A1
#  define OCR1x  OCR1A
#else
#  define NO_HARDWARE_PWM
#endif
```

krusli, 2018-01-11:
Edit 2: I have a feeling it's just a matter of the equivalent of `PWM_Init()` not being called on turning on the backlight LEDs:

```C
void timer1Init(void)
{
	// initialize timer 1
	timer1SetPrescaler( TIMER1PRESCALE );	// set prescaler
	outb(TCNT1H, 0);						// reset TCNT1
	outb(TCNT1L, 0);
	sbi(TIMSK, TOIE1);						// enable TCNT1 overflow
}
```

This is called before `timer1PWMBOn()`:
```C
void timer1PWMBOn(void)
{
	// turn on channel B (OC1B) PWM output
	// set OC1B as non-inverted PWM
/* http://www.nongnu.org/avr-libc/user-manual/group__avr__sfr.html
sbi(A, B) is A |= _BV(B) // BV: byte value
cbi(sfr, bit) is sfr &= ~(_BV(bit))
*/
	sbi(TCCR1A,COM1B1);
	cbi(TCCR1A,COM1B0);
}
```

After this `setPWM()` is called in ps2avrgb, which sets the value of the PWM (registers?), in which a block of code that seems to the corresponding code in QMK is also there in `void backlight_set()`

```C
__attribute__ ((weak))
void backlight_set(uint8_t level) {
  if (level > BACKLIGHT_LEVELS)
    level = BACKLIGHT_LEVELS;

  if (level == 0) {
    // Turn off PWM control on backlight pin
    TCCR1A &= ~(_BV(COM1x1));
  } else {
    // Turn on PWM control of backlight pin
    TCCR1A |= _BV(COM1x1);
  }
  // Set the brightness
  set_pwm(cie_lightness(TIMER_TOP * (uint32_t)level / BACKLIGHT_LEVELS));
}
```

ryanm101, 2018-01-11:
Tried B5, B6 and back to D4 again no luck.

The only time the backlights come on for me is during the flashing /bootloader processes.

ryanm101, 2018-01-11:
FYI: My repo with current settings etc
https://github.com/ryanm101/qmk_firmware/tree/PersonalLayout/keyboards/jj40

I'm using the layout "ryan"

krusli, 2018-01-11:
When you set it to say D4 do the LEDs come back on after the flash?

ryanm101, 2018-01-11:
No, mines not doing anything with Backlights..

Your underglow code works perfectly though

krusli, 2018-01-11:
What I meant was, when you flash the keyboard and the keyboard starts up for the first time, do the backlighting LEDs turn on? And could you control the brightness of the in-switch LEDs using the toggle keys while it's on, or is it essentially uncontrollable?

ryanm101, 2018-01-11:
No they dont turn on at all

FYI: https://github.com/qmk/qmk_firmware/pull/2253 (I you would rather do the PR, so it's against your name let me know and I will retract it.)

krusli, 2018-01-11:
I don't mind - go ahead with the PR.

And this is with all the settings? I'm just curious as the `fun40` keymap's README seems to note that at least the backlight LEDs would turn on after a keyboard flash, but fail to turn on again once the keyboard is rebooted.

From the README:
```
# Fun 40%.

* backlighting on flash  
  * backlight fails on reboot
```

ryanm101, 2018-01-11:
Let me diff him against my current branch just to be sure, i may have removed something during other tests

ryanm101, 2018-01-11:
ok i'm not sure how hes done it. I can see the new include of "backlight.h" in keymap.c and he calls backlight step. 

I tried to move them into the top level so it would apply to my code but it complains about backlight_step().

If i flash the Fun40 then backlight comes on and stays on. The keys has has as BL_ON and Step do nothing.

ryanm101, 2018-01-11:
ok

so adding
```
#ifdef BACKLIGHT_ENABLE
  #include "backlight.h"
#endif
```
to my jj40.h causes my backlights to stay on post flash however adding that causes the underglow to fail on unplug -> replugin

ryanm101, 2018-01-11:
If i leave plugged in after flash Backlights stay on but are uncontrollable. RGB stays on and is controllable...
Replugging Blacklights are off and uncontrollable. RGB works fine

krusli, 2018-01-11:
That's quite weird, `backlight.h` is just a bunch of function signatures that's already included when the keymap is compiled I think.

I'm working on trying to port over the code from the PS2AVRGB repo right now.

ryanm101, 2018-01-11:
This is wired. Now backlights are not working after flash.. :/

krusli, 2018-01-11:
Try this out? [GitHub gist](https://gist.github.com/krusli/50eb417c20db1989784dbf8b44a6d225)

You'll have to modify `quantum.c` and add `__attribute__ ((weak))` above the declaration of `void backlight_task(void)` to let my custom function in `jj40.c` override it.

Hopefully I didn't miss any initialisation code needed for the backlight LEDs to at least work. QMK still compiles just fine with these changes at least.

The code in the gist is still quite minimal - I'm not even sure if I've gotten all the necessary initialisation for the backlighting, or if I have to do some sort of preparation or anything before issuing commands and modifying the contents of the AVR registers. It's also stuck at a fixed brightness level (if it even works) and does not check whether backlighting is turned on or not I think for now.

ryanm101, 2018-01-11:
Ok so build compiles and flashes successfully
Backlight state after flash: Off
Tried keycodes:   BL_TOGG, BL_ON, BL_OFF, BL_INC  *no effect*

krusli, 2018-01-11:
Add this to `backlight_task(void)` in `jj40.c`:

```C
// turn on LED
PORTD |= (1 << 4);  // PIN D4
```

ryanm101, 2018-01-11:
no change

ryanm101, 2018-01-11:
Repo with changes: https://github.com/ryanm101/qmk_firmware/tree/PersonalLayout
Just in case you want to validate my changes

krusli, 2018-01-11:
Have a look at this [GeekHack thread](https://geekhack.org/index.php?topic=85943.msg2312974#msg2312974)

It seems that someone got RGB LEDs working on the b.face X2, which is also a PS2AVRGB board. I'll try to have a look tomorrow after I catch some shuteye for now.

ryanm101, 2018-01-11:
Ahh. No worries thanks for looking into this. It wouldn't bother me so much except I went to all the trouble of soldering those LEDs  :)

krusli, 2018-01-11:
I'll have a look at #1061. Looks like someone got Caps/Scroll Lock LEDs working on the PS2AVRGB branch. I'll have a look to see if something similar to that would work.

ryanm101, 2018-01-11:
I just copied and pasted that in wholesale. 

Nope didnt work :(

krusli, 2018-01-11:
Did you change the pin back to D4 (just in case)?

Or in the mean time, could you do this (hacky) thing as in [this commit](https://github.com/ekx/qmk_firmware/blob/bface/keyboards/ps2avrGB/ps2avrGB.c) from the issue I linked above? 

```
void led_set_user(uint8_t usb_led) { 
    if (usb_led & (1 << USB_LED_NUM_LOCK)) {
        DDRD &= ~(1 << 0); PORTD &= ~(1 << 0);
    } else {
        DDRD |= (1 << 0); PORTD |= (1 << 0); 
    }
        
    if (usb_led & (1 << USB_LED_CAPS_LOCK)) {
        DDRD &= ~(1 << 4); PORTD &= ~(1 << 4); // originally (1 << 1), port/pin D1
    } else {
        DDRD |= (1 << 4); PORTD |= (1 << 4);
    }
        
    if (usb_led & (1 << USB_LED_SCROLL_LOCK)) {
        DDRD &= ~(1 << 6); PORTD &= ~(1 << 6);      
    } else {
        DDRD |= (1 << 6); PORTD |= (1 << 6);     
    }
}
```

And try to toggle Caps Lock on and off and see if it toggles the backlight.

ryanm101, 2018-01-11:
Yes pin is D4 (you can see my exact config in the repo i linked above (using the keymap "ryan")

I'll try that now.

ryanm101, 2018-01-11:
Soo that'svery close..

When I press my LAYER keys with caps lock OFF the backlights flicker quickly on->off 
When I press my LAYER keys with caps lock ON nothing happens
However If I hold LAYER and toggle caps ON then nothing but when toggling caps OFF the backlights flicker on->off

krusli, 2018-01-11:
That's great to hear.

I'll be signing off for now but I'll see if I can trace through the code and see if I could figure it out later. It might just be the LEDs not playing nice with the soft PWM of the existing backlighting code, or it might just be because it only activates on getting the signal.

This is especially interesting as having the code in `matrix_scan_user` should, in theory, mean that it is executed at each matrix scan but it didn't in fact turn the backlight on in your case. Maybe there's some initialisation happening behind the scenes?

I could order some LEDs and solder them onto my board. I'm actually not sure of this but do you think installing just 1 LED would work? I would rather have the ease of doing a switch swap if necessary than to have backlighting on my board as LEDs are annoying to desolder for me. A quick Google search seems to support this theory (boards can have several LEDs fail while the others stay on) so I might just give this a shot.

ryanm101, 2018-01-11:
I've pushed all my changes to my "PersonalLayout" branch mentioned above so you can make sure i've done it correctly

ryanm101, 2018-01-11:
ok SO I am an idiot.. I'd left `DDRD &= ~(1 << 4); PORTD &= ~(1 << 4);
` in the matrix_scan_user

Removing it had had the Following Effect
Flash Board -> LEDs: OFF
Caps Lock OFF: Press LAYER KEY key LEDs turn on
Toggling caps lock (via main laptop KB or via LAYER+CAPSLOCK) toggles LEDs on/off
Does not remember state between unplug/replug

Behaviour 1:
Using my laptop keyboard:
Plugin JJ40 -> LEDs off
on main KB turn on CapsLock -> Nothing
then
on main KB turn off CapsLock -> LEDs ON
then 
on main KB turn off CapsLock ->LEDs OFF

Behaviour 2:
Using my laptop keyboard:
Plugin JJ40 -> LEDs off
Press LAYER on JJ40 -> LEDs ON
on main KB turn on CapsLock -> LEDs OFF
then
on main KB turn off CapsLock -> LEDs ON
then 
on main KB turn on CapsLock ->LEDs OFF

No idea why the keyboard on my laptop is able to control the LEDs on the JJ40 :/

sanosis, 2018-01-11:
On my jj40 (pcb 1.5) with fun40 i get:
+ bootloader flashing leds
+ after flashing leds are on
+ i can turn the leds on/off bl_togg
+ bl_inc/dec don't change brightness but will sometimes turn ledds on/off

krusli, 2018-01-11:
Is this straight from the repo? Does BL_INC and BL_DEC keep "working" after an unplug and replug? What about BL_TOG?

Worst goes to worst I'll just buy and solder a few LEDs myself so I could more effectively test the code I'm working on.

sanosis, 2018-01-12:
Unplug kills the leds - none of the switches I've tried turned them on.

krusli, 2018-01-12:
So it seems to just be missing some initialisation code.

Interestingly using the backlight LEDs as a "Caps Lock" LED seems to somewhat work: see the `led_set_user()` block above.

I'm thinking that perhaps some of the AVR registers aren't set properly by QMK for the initialisation of the LEDs.

Maybe you could try dropping the `DDRD &= ~(1 << 4); PORTD &= ~(1 << 4);` line to turn on the LEDs and the `DDRD |= (1 << 4); PORTD |= (1 << 4);` line to turn off the LEDs in the backlighting functions in either `quantum.c` or `backlight.c` and see what happens? We could then just move the changes out to functions in `jj40.c` to override the behavior instead of modifying the QMK quantum core code.

krusli, 2018-01-12:
@ryanm101 if you're using Windows I think [caps lock state is shared between keyboards](https://superuser.com/questions/858872/why-two-different-keyboard-connected-to-a-single-computer-have-telepathy). Activating caps lock on your laptop also sends the USB caps lock signal to your JJ40, which then triggers the LED enabling code.

The JJ40 (and the other PS2AVRGB) boards use PWM implemented using timer1 in PS2AVRGB to adjust the brightness of the LEDs. PWM (pulse width modulation) is basically reducing the intensity by (uniformly and rapidly) turning the LEDs on and off such that they're, say, on only half of the time in one time period for half the intensity.

QMK should already have code for PWM - either hardware PWM for the pins B5-B7 or soft PWM for the other pins (simulated PWM?), although I'm not sure if it even works to adjust the brightness while it is on on the JJ40.

sanosis, 2018-01-12:
@krusli - my guess is that you could use just one led and you can use unpopulated key (mit/grid) and put it under the bord. You  don't even have to solder that one, just bend the legs (not very practical for long term use, but works for testing).
You can use a multimeter instead of an actual led (but it's less convenient on the desk).

krusli, 2018-01-12:
@sanosis literally just left the electronics store with 1 LED as you sent me this. Thanks for the tip on just bending the legs - I did ask the mech keys discord here on it and it seemed to confirm my suspicion that that would work. Also I actually did desolder a board with LEDs once and the switches with LEDs still on them still worked even with missing LEDs in the circuit.

ryanm101, 2018-01-12:
@krusli Using Fedora.  I'll be unable to test till I get home from work which will likely be after your offline but if you post code etc to try i'll get back to you asap

krusli, 2018-01-12:
Got the single LED working! I'll work on making my custom `backlight.c` file for the board now.

krusli, 2018-01-12:
I'm still working on it - if you want to follow along with the progress have a look at my [repo fork](https://github.com/krusli/qmk_firmware/)

krusli, 2018-01-12:
I got `BL_TOGG` working in [this commit](https://github.com/krusli/qmk_firmware/commit/5396f58da3575a6eb77bad99a465dc703c203473). Somehow I couldn't override the functions in `quantum.c` even after using a `#ifndef`, and yes I've updated the Makefile rules for it.

I'm going to focus on getting it to work for now. Cleanup comes later.

krusli, 2018-01-12:
The reason you're able to turn the LEDs off with `BL_STEP` is because there's a check that turns off the LEDs when the brightness level of the LEDs is set to 0 by the way.

krusli, 2018-01-12:
Bingo, I'm done. Check out a copy of my repo [here](https://github.com/krusli/qmk_firmware) and you should now be able to fully use `BL_TOGG` and `BL_STEP`. `BL_BRTG` isn't working yet as I haven't ported over the fading in/out code but it shouldn't be that bad now that I have everything working.

krusli, 2018-01-12:
Breathing is working now. Just `#define BACKLIGHT_BREATHING` in `config.h` if you want to use it.

ryanm101, 2018-01-12:
Nice i'll give it a whirl tonight.. good job

krusli, 2018-01-12:
Backlighting should be working already. @ryanm101 could you do a clone of my repo and see if everything works on the backlighting?

I did notice one tiny issue - with breathing turned on the backlight can't be turned off, but once it is disabled (with the keycode) it functions as normal again.

danielo515, 2018-01-12:
Are different modes supposed to work ? They are not working for me, and after activating one of them the underglow has stopped working. What a pitty

ryanm101, 2018-01-12:
@krusli 

I copied my Keymap in for testing:

LED State after 1st flash: Off
BL_TOGG: Toggles ON but not OFF, results in Breathing effect by default
BL_ON: Will put on but still breathing
BL_OFF: Turns off as expected

Commenting out #define Backlight_breathing in config.h
BL_TOGG: Works as expected
BL_ON: Works
BL_OFF: Works
BL_INC: Works

RGB_TOG:Works

krusli, 2018-01-12:
@danielo515 did you check out the latest version? The underglow would get stuck because I forgot to call `rgblight_task()` in `matrix_scan_user()`.

@ryanm101 the [documentation](https://docs.qmk.fm/feature_backlight.html) does say that it should be defined in config.h and I'd like to keep it consistent with that.

ryanm101, 2018-01-13:
`in your config.h`
So maybe remove it from the `config.h` and let people enable it by placing it in the one in their keymap folder, the same as the prevent_stuck_modifiers

[7 more replies]

closed 2018-01-13 by ryanm101