qmk/qmk_firmware #2226: Backlight flashing on xd60 after #2187

michaeldauria, 2018-01-05:
After flashing #2187, my backlight constantly flashes while on on my xd64. Once I revert the offending commit, everything works as expected.

I tried playing around with some settings, but ultimately I could not get it to work properly.

Thanks,

Michael

JonasGe, 2018-01-17:
I am having the same issue on my XD75.

The backlight LEDs are on pin F5. Decreasing the BL level with BL_DEC eventually turns off the LEDs. BL_TOG and BL_INC don't do anything other than turning the LEDs back on. BL_INC isn't supposed to do any dimming/brightening as pin F5 of the ATMega32u4 is no PWM pin but I suppose BL_TOG should also be able to turn the LEDs off?

krohmag, 2018-01-20:
Same issue on my xd75 as well. Resetting back to commit d6215ad6aff3857cea8a6877b56a547a76ab13ac, the commit before #2187, resolved my issue as well, but that didn't satisfy me. 

Going through #2187, the issue is line 907 in `quantum/quantum.c` - order of operations wasn't followed like it was in the previous commit. 
The line was previously `backlight_tick = (backlight_tick + 1) % 16;` but #2187 drops the parenthesis. Essentially, the issue comes down to bad math. After adding the parenthesis back in, I can run off of master without an issue.

kaylanm, 2018-01-23:
This definitely fixes the immediate flashing problem but there is still some flicker when keying on my XD75. Is anyone else seeing that behavior? TMK does not have the same issue on the same hardware.

skullydazed, 2018-02-01:
This looks like a simple patch. Would someone like to put a PR together and we'll get it merged in?

PR #2329 (2018-02-02): Add missing parenthesis back to backlight_tick to fix #2226
xd60 and xd75 keyboards (others using soft PWM as well?) experience backlight flashing when this math is incorrect.

Fixes #2226.
closed 2018-02-02 by jackhumbert