qmk/qmk_firmware #2229: [BUG] Adafruit BLE 32u4 OUT_AUTO not applied by default

harshitgoel96, 2018-01-05:
Hi 
I am using Adafruit Bluefruit 32u4 on my board, with code setup provided in MF68_BLE. 
When I connect the bluetooth and USB on first boot, the board is outputting on both USB and BT at the same time. This results in double character input from the board.

I have added keycode to add OUT_AUTO on my keymap. Once this is pressed, this issue is resolved, and i am able to switch between BT and USB seamlessly. 

How can we fix this ?

colemarkham, 2018-02-12:
I know that KC_RESET worked last time I tested it. You have to #define CATERINA_BOOTLOADER like on the pro micro, but even that isn't working now. Something must have changed between Sept/Oct and now. I'm currently not getting anything to work on my Feather BLE 32u4 with code that worked a few months ago. I do remember something about needing an older firmware version for the RF module, so I'm looking for that right now.

harshitgoel96, 2018-02-13:
@colemarkham thanks for looking into this. I can confirm that I am using the latest firmware that was available in Aug 2017. I completed the build in December and that is when I noticed the issue.

I have not tried using  #define CATERINA_BOOTLOADER  option. i will try to this today and let you know if this changes anything.

drashna, 2018-03-25:
There was a big update to the bootloader code and make process. 

Try updating QMK and see if that helps. 

Also, CATERINA won't help unless you're using the caterina bootloader (such as the pro micro). If you're using DFU, then ... "nothing will happen" basically.

harshitgoel96, 2018-03-26:
@drashna first of all, thanks a lot for trying to help me with my so many queries.

As per few different post, it looks like CATERINA is the default bootloader on Feather 32u4, thought i am not sure.
I have not been able to test it yet.

drashna, 2018-03-26:
Welcome. I've recently been promoted to a collaborator, so I went through the entire backlog of issues (no small task, let me tell you!). 

For the Feather 32U4, if it's this:
https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/overview

It does require a double-tap to enter bootloader, which is a distinctly CATERINA thing, IIRC. 
And the newer bootloader options, you want to set the "BOOTLOADER" to caterina, as per: https://docs.qmk.fm/config_options.html#avr-mcu-options

This way, it picks up the caterina specific code, and does proper detection of file size.

harshitgoel96, 2018-05-05:
So i have tested the Bootloader option and now I can enter the DFU mode without any issues.
Closing the defect, will open a PR soon.

closed 2018-05-05 by harshitgoel96