qmk/qmk_firmware #2234: Proposal: Predefine compose macros for non-ASCII characters

rickard-von-essen, 2018-01-06:
## Background

I'm a programmer and I mainly use a US ANSI (QWERTY) layout since this is good for accessing special characters. But since I'm also writing in Swedish from time to time I like quick access to the Swedish extra characters Å, Ä, and Ö without switching (software) keymap.

## Suggestion

This can easily be achieved with AltGr (win), Opt (macOS), and Compose (Linux) macros. Since this is probably a very common situation and many implemented this in there own custom `keymap.c` I think this should be a generic feature QMK provided ready to be used.

Where you can use a keycode to send say _Å_ with `C_SA_AA` and it would use the correct macro depending on your platform set by something similar to `set_unicode_input_mode`.

Keycode | Char | Win | macOS | Linux
---------|------|-----|---------|------
`C_SE_AA`  |    Å    | AltGr + w | Opt + a  | Compose, o, a
`C_SE_AE`  |    Ä    | AltGr + q | Opt + u, a | Compose, ", a
`C_SE_OSLH` |   Ö    | AltGr + p | Opt + u, o | Compose, ", o

If this sounds good, I can start working on a PR adding this.

A few question:

1. Should this be a separate feature or should this be squeezed into the Unicode support?
1. Naming.. If this is a separate feature, what should it be called? Maybe _Compose Macros_ or something similar.
1. Which prefix for keycodes should be used? I suggest `C_` and that they are named the same way as in the language specific keymaps (i.e. `C_SE_AA`).

fredizzimo, 2018-04-07:
I'm not sure if this needs a feature of it's own, since it's relatively simple to implement by custom keycodes and `SEND_STRING` in your own keymap. 

But if you find a way to make it into a feature that can be customizable to support all combinations, then that would be great. I don't think it would be possible to combine with the unicode, so it would have to be a spearate feature. 

For keycodes, I think we have to let the users define their own keycodes for this, since there are so many languages and combinations, and we don't want to waste memory for things that are not used in that specific language.

drashna, 2018-10-22:
Also, unicode could be used to handle this, as well. 

Additionally, the US Intl on Windows will handle this relatively well, too.

closed 2018-10-22 by drashna