qmk/qmk_firmware #2237: creating layout from xkb layout

plexigras, 2018-01-07:
a xkb layout definition looks something like this

```
    key <AE01> {	[	  1,	exclam 		]	};
    key <AE02> {	[	  2,	at		]	};
    key <AE03> {	[	  3,	numbersign	]	};
    key <AE04> {	[	  4,	dollar		]	};
    key <AE05> {	[	  5,	percent		]	};
    key <AE06> {	[	  6,	asciicircum	]	};
    key <AE07> {	[	  7,	ampersand	]	};
    key <AE08> {	[	  8,	asterisk	]	};
    ...
```

i would like to have a key that sends `asterisk` in the form of `Shift+AE08`

maybe even cooler would be if i could have a second layout like
```
    key <AE08> {	[	asterisk,	8	]	};
    ...
```
that makes it so if i press `AE08` it sends `Shift+AE08` and if i press `Shift+AE08` it sends just `AE08`

then i could easily have a layout that is different then the layout of the keyboard.

drashna, 2018-03-25:
Ah, so it inverts the Shifted state, based on the keypresses and keycode, correct?

drashna, 2018-10-22:
You could do this with `process_record_user` actually. 

Check the mod state, clear it, and then set weak mods, based on the mod state.  

That would get the behavior that you want, and would be probably 5-6 lines of code.

plexigras, 2018-10-22:
https://gist.github.com/plexigras/5ed23fc91feb318d8d410aa0a4d68883

the `layout.txt` file contains the output of `xmodmap -pke`

jackhumbert, 2020-03-17:
It sounds like you got this implemented - if you have any code to share, or additional complications, feel free to drop them here!

closed 2020-03-17 by jackhumbert