qmk/qmk_firmware #2238: Question: Using mod tap, but sometimes I get a hold when I intend to get a press

JakeSummers, 2018-01-07:
Good Morning!

I am using the [ModTap](https://docs.qmk.fm/feature_advanced_keycodes.html#mod-tap) functionality to make the 'B' key a 'B' when tapped but a 'CTRL' key when held.  [See here](https://github.com/JakeSummers/qmk_firmware/blob/44987622e0ededfc2190f105073b3efe23755463/keyboards/preonic/keymaps/jake/keymap.c#L77).

This for the most part has been working fairly well.   Unfortunately, sometimes I am getting a control when I am expecting to get a 'b'.  This can result in odd behaviour like pressing CTRL-A.

Is there a timeout that I could tweak to help address this issue?

Thanks!  If there is somewhere better that I should post this question, please let me know.

Jake

danielo515, 2018-01-12:
You should try to adjust the TAPPING_TERM timeout

```
#define TAPPING_TERM 200
how long before a tap becomes a hold
```

drashna, 2018-03-25:
Alternatively, try add `#define RETRO_TAPPING` to your keymap's `config.h`, as this will send `b` if nothing else is hit while it's being treated as a hold.

lindhe, 2018-07-08:
What's the default value of TAPPING_TERM?

drashna, 2018-07-08:
If you add the following to your keymap's config.h file, it will set it for just your keymap: 
```c
#ifdef TAPPING_TERM
#undef TAPPING_TERM
#endif // TAPPING_TERM
#define TAPPING_TERM 175
```

drashna, 2018-07-08:
No, there isn't.   I mean, there is, but .. it doesn't work right (and is a PR, not in the official QMK repo). 

So if you create a custom keycode for this, you could. 

Though ...
> Is there a way to have different timeouts for different keys? I'd like my Enter key to be slower than the Shift and Ctrl keys. > Sorry for being picky.

Do you mean for how often it repeats?   Or for stuff like Mod Tap? 

Also, for mod tap, adding the following to your config.h file may help with how they trigger:
```c
#define IGNORE_MOD_TAP_INTERRUPT
```

drashna, 2018-10-22:
Also, you should check out these options, too:
https://docs.qmk.fm/#/feature_advanced_keycodes?id=permissive-hold

And it sounds like this may have been resolved, or abandoned.
If not, then please reopen.

closed 2018-10-22 by drashna