qmk/qmk_firmware #2241: [Feature Suggestion] Emacs style `C-u n` for sending key or macro multiple times

kennycoc, 2018-01-08:
It think it would be useful to have functionality similar to the emacs 'universal argument' feature.
This would need a key, or key combo defined as a 'multi_key'

The way this works in emacs, I press the key combo, any number of digits, and then another key or command.
e.g. `C-u 16 down_key` == send the down key sixteen times.

It won't be able to work exactly like this, since this would disallow sending a number key multiple times.. maybe it should only count numbers while the multi_key is held down?
That may cause issues if the numbers are on a different layer though; not sure. I'll probably need to do some experimenting to find the best way to implement it.

I'd be happy to contribute the code if there is any community interest, but I won't be able to begin work till I get my planck (in about a month :( ).

skullydazed, 2018-02-01:
Interestingly enough @mavanmanen and I were talking about a feature like this on [gitter](https://gitter.im/qmk/qmk_firmware) this morning. This would definitely be a good feature to have.

One of the issues I thought about with an approach like you describe is that it would be difficult to support mods like Ctrl+V. That was the specific use case that prompted the conversation this morning.

kennycoc, 2018-02-01:
The rest of my parts should be here in a few days.. I've been thinking about this a bit more since then; here's my current idea of how it could work:
Type any number n while holding multi-key to set number of times to repeat.
Release multi-key and type any arbitrary amount of text or commands.
tap multi-key to finalize and send keystrokes n times.

This may be overkill for simple use cases though, and not sure yet how restrictive the memory will be.

mavanmanen, 2018-02-02:
This is my current way of implementing functionality like this: https://pastebin.com/C1P6sqM1

mavanmanen, 2018-02-04:
Second iteration, using a callback to call after input has been given, allows input of numbers and letters + space: https://gist.github.com/mavanmanen/3a919d4fc8f6e84aeda33442a5fc9261

jackhumbert, 2020-03-17:
This sounds like a pretty good implementation - if you have any additional complications with it, feel free to drop them here!

closed 2020-03-17 by jackhumbert