qmk/qmk_firmware #2242: OSL(layer) doesn't return to base after pressing a macro key (SEND_STRING variant)

Jipsu, 2018-01-08:
Oneshot layers don't seem to unregister when a SEND_STRING macro key is pressed. Same issue with RGB_TOG key. Basic keys and modified keys like LCTL(KC_C) work as expected, returning me to base layer. Is there something I have overlooked? I've set macros in keymap.c and enums in kb.h here: https://github.com/Jipsu/qmk_firmware/tree/master/keyboards/shortcut

drashna, 2018-01-08:
For the macros, because of how they're handled, this may be "normal", and there isn't a good way to handle it, IIRC.

But you can manually call `reset_oneshot_layer()` in your macro, to fix this. 

As for the ROG_TOG, add it to your `process_record_user`, and add the above function. But instead of "return false", use "return true" instead.

Also, I would recommend `layer_state_set_user` to set the RGB, as this is called every time the layer is switched, so you don't have to rely on matrix scan.

Jipsu, 2018-01-10:
Thanks for guidance! I wasn't able to get `reset_oneshot_layer()` to work but I used `layer_off (layer)` instead. I might look into calling that function later with better time.

I looked into your RGB code, seems it plays an effect at layer change and then applies colour based on current default layer. Definitely more powerful than my current implementation. I'll have to either change to DF() or make the code read current layer instead. Food for thought.

closed 2018-01-11 by Jipsu