qmk/qmk_firmware #2248: NKRO does not work after OS reboot/start on certain setups

raetiacorvus, 2018-01-10:
Reconnecting the keyboard by plugging it out and back in results in NKRO working as expected. E.g. it honors nkro eeprom setting and reacts correctly by toggling it with MAGIC_TOGGLE_NKRO (`keymap_config.nkro`).

I was able to observe this behavior both on Windows and Arch Linux on my desktop computer as well as on arch linux on my Macbook Air. It works fine with OSX without reconnecting the board on the same Macbook Air and on Windows on my brothers laptop. Both my Katana60 as well as a Arduino Pro Micro based board I am working on show the same behavior.

Using <kbd>magic key</kbd> + <kbd>s</kbd> (and <kbd>magic key</kbd> + <kbd>e</kbd> ) with `hid_listen` shows that `keymap_config.nkro` and the eeprom equivalent get set correctly but NKRO just does not work in any tester till the keyboard is restarted in some way e.g. plugging it out and back in or flashing a new firmware.

drashna, 2018-03-25:
After updating the repo, does this still happen?

raetiacorvus, 2018-04-08:
Sorry for the delay in response but yes I did some testing with `0.6.1` and the problem still persists.

It looks like that @fredizzimo is right. `keyboard_protocol` stays `0` on a reboot and only is set to `1` on a replug.
Forcing it to `1` like suggested in the following line results in working nkro.
https://github.com/qmk/qmk_firmware/blob/1f778684276e90609c183045a72c96c3c12d452c/tmk_core/protocol/lufa/lufa.c#L554

dsvensson, 2022-01-18:
I have this same issue with my Infinity Ergodox. My motherboard is MSI MAG B550 TOMAHAWK with latest BIOS if that matters. The problem did not happen with my old computer iirc. Tried various USB ports hoping there's some controller variance, but no success. It's not OS related either, happens with both Windows and Linux.

jakemauer, 2022-05-04:
So I'm resurrecting this issue as I came here via a Google search. I had this problem with my Massdrop Alt, even with FORCE_NKRO enabled it never worked after boot. I would always have to toggle it on. I tried to make adjustments to the usb C files but the arm files that TMK uses in the Massdrop builds already have keyboard_protocol set to 1.

On a hunch I enabled "Legacy USB support" in my BIOS and it fixed it! NKRO is enabled by default on boot now. For reference I'm on a Gigabyte Aorus Pro Z390 board with an EFI bios.

zvecr, 2022-08-29:
This issue has been automatically closed because it has not had any recent activity.
If this issue is still valid, re-open the issue and let us know.

closed 2022-08-29 by zvecr

IlyasYOY, 2023-06-14:
Update: because of this option being on Caps Lock doesn't seem to change input source. Everything works fine if I have NKRO off.

FabienFellay, 2024-04-30:
If Google brought you here and the above seems like the issue you have, please, go to issue #23255 where you might find a solution.