qmk/qmk_firmware #2249: €30 bounty for implementing customisable space cadet keys

Krakob, 2018-01-10:
Followup from #1884:
I'm offering €30 to anyone who can implement a simple way to make localised space cadet keys:

* User is given a simple way to define a generalised space cadet key in the form (or similar to): `#define my_pseudokeycode cadet(key_on_hold, mod_on_tap, key_on_tap)`. Example usage: `#define LCTL cadet(KC_LCTL, MOD_RALT, KC_7)` would let the user bind a key to `LCTL`, which would produce left control when held, and AltGr + 7 (which is `{` on a Swedish keyboard).
* Function must support AltGr (not the same as ctrl + alt)!
* An alternative form for layers, using MO()-style held layers should be available as well. E.g. `#define my_layerkeycode cadet_layer(L_TARGET, MOD_RALT, KC_7)` should switch to L_TARGET on hold and produce AltGr + 7 on tap.
* User should not have to implement their own functions or such for this. Using it should consist of a simple definition of the key and then binding it in the keymap.

Bounty can be paid via PayPal (international) or Swish (Sweden). Conditions are that this feature is merged into QMK and I'm personally able to verify its functionality.

TurboMech, 2018-01-10:
You're trying to have two different things on a tap referenced.  I don't think thats going to work very well, if at all possible.  You could attempt this with `tap_dance`, maybe combining `tap_dance` and `MOD_TAP` this could be theoretically possible.  If I have time I'll look into it this weekend.

Krakob, 2018-01-10:
It's functional - see the linked issue's last post ([or just the source code](https://gitlab.com/Krakob/qmk_firmware/blob/krakob-xd60/keyboards/xd60/keymaps/krakob/keymap.c)). This is for implementing a simpler way to do it for the benefit of all ;)

Krakob, 2018-03-25:
Mod taps don't work nicely with modifiers on the tap keys, so not quite.

drashna, 2018-03-25:
Ah, yeah.  That's a seperate issue with how many of the macros/aliases work.  Which needs to be fixed.

drashna, 2019-03-29:
@Krakob I think this PR covers what you want: https://github.com/qmk/qmk_firmware/pull/5277

closed 2019-11-21 by drashna