qmk/qmk_firmware #2271: Can not compile hex

Benjamin-Blackswan, 2018-01-14:
I don't know if this is the right place to post this question, but I am not getting any response from reddit.

I have having trouble compiling hex file for Ergodox Infinity.

Getting following error $ make ergodox_infinity:default

quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: No such file or directory make[1]: *** No rule to make target 'lib/ugfx/gfx.mk'. Stop. Make finished with errors make: *** [Makefile:543: ergodox_infinity:default] Error 1 bash: $: command not found

I have no issue compiling some other keyboard such as planck:default

Any help would be great.

shelaf, 2018-01-15:
If you download source code from github as a zip file, you don't have 'qmk_firmware/lib/ugfx' directory.
Please use git clone command to download it.

drashna, 2018-01-15:
Running "make git-submodule" should fix this, IIRC.

skullydazed, 2018-02-01:
@Benjamin-Blackswan did `make git-submodule` fix this for you, as @drashna suggested? If so can we close this?

Benjamin-Blackswan, 2018-02-03:
No, it got this error

$ make git-submodule
': not a valid identifierline 88: export: `dashless
fatal: Not a git repository (or any of the parent directories): .git
make: *** [Makefile:579: git-submodule] Error 128

loadedsith, 2018-03-01:
That sounds like there's no git repo, did you download as a zip? You're going to want to download the repo with git instead.

drashna, 2018-03-25:
Did you get this working?

dorton, 2018-10-20:
i had the same error and `make git-submodule` was the solution. Thanks!