qmk/qmk_firmware #2274: Is it possible to use Bluefruit LE UART with an arduino pro micro?

steinerlein, 2018-01-15:
I am looking to hook [this](https://www.adafruit.com/product/2479) up to a pro micro (32u4)
So far, I had no luck getting any output over the serial port.
Is it possible? If so, how?
Thanks! QMK is awesome 🥇

steinerlein, 2018-01-16:
After having spent some more time on this, I have been successful on my own. For documentation purposes here is what I did:
In order to use the 
```
BLUETOOTH =AdafruitBLE 
```
option I recreated the 32u4 feather setup:
1. I flashed the firmware for the bluefruit SPI friend onto the nrf module. I have a bluefruit uart friend, but the ble module is the same. Follow the Adafruit guidelines for this. 
2. I soldered six wires between a pro micro (32u4) and the nrf module directly. Follow the adafruit schematics for the feather 32u4. 
Connections are SPI, power and some for resetting the nrf module. 
3. Enabled the aforementioned option in the rules.mk of my board and connected it to my phone as a test. 

This is still missing power, but that's next. 
Will post some pics tomorrow

wez, 2018-01-17:
FWIW, https://github.com/wez/qmk_firmware/commits/flutterby is the branch in my fork for the `flutterby` keyboard that I built that uses the SPIFRIEND, which is built on top of the Feather32u4 BLE code base where I prototyped the baseline support.

I also wrote up a build log here, that includes a schematic:
https://www.facebook.com/notes/wez-furlong/building-a-keyboard-ii/10154296367151787/

A couple of problems I found when integrating against the BLE modules from adafruit:

* Depending on the firmware version, the BLE advertising intervals were too big and that resulted in a very laggy typing experience.  IIRC I commented somewhere in the code which version worked for me.
* If you send SDEP packets to the hardware faster than the Central (your computer) drains them from the BLE module, you can overflow it and cause it to reset

steinerlein, 2018-01-17:
Awesome, thanks for the feedback! Will check it out and see if the firmware update helps.

steinerlein, 2018-01-18:
@wez The firmware downgrade did improve latency a lot! Thanks for the tip. Can you make any sense of the problem with pressing multiple keys (see further up)

wez, 2018-01-18:
Sounds like something might be explicitly clearing the keyboard and sending the cleared state to the spifriend each time the matrix is changing?  Hard to say.

targetaw, 2018-05-12:
Using the Bluetooth = AdafruitBLE option worked great for me as well. My question is, since I haven't seen this in the qmk.fm documentation anywhere, where could I have learned of this option if it weren't for this thread?

In case you haven't guessed, programming isn't my strong point so thank you in advance for answering what may be a very basic question.

For the record, this is the bluetooth numpad I have built as a stepping stone to a full sized keyboard.
[image]

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc