qmk/qmk_firmware #2277: Tab dance doesn't work on ymd96

underpk, 2018-01-16:
I tried to add the tap dance following QMK document but no matter how I try it always error.

Compiling: keyboards/ymd96/keymaps/default/keymap.c                                                keyboards/ymd96/keymaps/default/keymap.c:34:1: error: unknown type name 'qk_tap_dance_action_t'
 qk_tap_dance_action_t tap_dance_actions[] = {
 ^
keyboards/ymd96/keymaps/default/keymap.c:35:3: error: implicit declaration of function 'ACTION_TAP_DANCE_DOUBLE' [-Werror=implicit-function-declaration]

https://github.com/qmk/qmk_firmware/tree/master/keyboards/ymd96

drashna, 2018-01-16:
You need `TAP_DANCE_ENABLE = yes` in your rules.mk file. 

https://docs.qmk.fm/feature_tap_dance.html#simple-example

underpk, 2018-01-16:
Hi, thank for your help. Here is what my keymap.c currently look like, at first I thought it was my config that make things go wrong so I download a fresh copy of YMD96 and just adding tap dance but it still give the same error. 

keymap.c
https://pastebin.com/h2hhLGsD

rules.mk
https://pastebin.com/z8VgzFhE

the error
https://pastebin.com/2mPW1xsY

I try adding tap dance in other keyboard (tada68) with the same config and there is no error.
So my guess is there is probably something wrong with this keyboard config itself?

drashna, 2018-01-16:
Looks like this keyboard needs `#include "quantum.h"` in the keymap.c file.  Add that, and it should compile.

underpk, 2018-01-16:
Man I try fixing this error for whole day without a clue what is going on and you help me fix it within a min!

Thank you so much!

drashna, 2018-01-16:
Well, most if the keyboards I've used don't need this. So it may be how the keyboard is set up.  

Either way, glad to have helped!

underpk, 2018-01-16:
Hi I think there is a few bug here
I follow the **"Example 1: Send : on Single Tap, ; on Double Tap"**
there is no error when I compile but the key won't work if I'm not pressing other button first.

I mean if I just press " TD(CT_CLN)" it does nothing, the ":" or ";" will only appear after I press other button first. 

If this is normal? I never use a keyboard with this QMK firmware before.

drashna, 2018-01-16:
Create a "config.h" file, with the following contents: 

```
#ifndef CONFIG_USER_H
#define CONFIG_USER_H

#include QMK_KEYBOARD_CONFIG_H

#define TAPPING_TERM 200

#endif
```

The key being the "tapping term" variable.

underpk, 2018-01-16:
I tried and it doesn't fix the problem

I read this thread https://github.com/qmk/qmk_firmware/issues/2162 and seems like @danielo515 also have the same problem as me the keyboard also use the same ATmega32a chip.

anyway thank you for taking your time again.

krusli, 2018-01-29:
Go to `matrix.c` and add `matrix_scan_quantum()` within `uint8_t matrix_scan(void)`.

Have a look at the JJ40 `matrix.c` file [here](https://github.com/qmk/qmk_firmware/blob/master/keyboards/jj40/matrix.c).

krusli, 2018-01-29:
Add `matrix_scan_quantum()` before `return 1` in `uint8_t matrix_scan(void)`.

The reason Tap Dance does not seem to be working is because QMK's matrix scanning code, which then calls `matrix_scan_tap_dance()` which handles tap dance keys' timeouts isn't called by the custom matrix scanning code there.

I noticed and fixed this when porting backlighting from the PS2AVRGB code for the JJ40. If anyone is interested in replicating/testing that feature too with this board (or any other PS2AVRGB boards), have a look at the [JJ40 repo code](https://github.com/qmk/qmk_firmware/blob/master/keyboards/jj40/).

underpk, 2018-01-29:
Thank you so much @krusli I would try it out once I back home. 👍  👍  👍

underpk, 2018-01-29:
Hi @krusli  I tested and it does error...

```
Compiling: keyboards/ymd96/keymaps/default/keymap.c                                                 [OK]
Linking: .build/ymd96_default.elf                                                                   [ERRORS]
 |
 | .build/obj_ymd96_default/quantum/quantum.o: In function `matrix_scan_quantum':
 | C:\qmk_firmware/quantum/quantum.c:1192: undefined reference to `matrix_scan_kb'
 | collect2.exe: error: ld returned 1 exit status
 |

```

I don't know what to do next...

krusli, 2018-01-29:
Use ymd96.c from my [GitHub gist](https://gist.github.com/krusli/c5dbaff00985389e274522678cf69331). The code is pretty much identical to the JJ40 code.

underpk, 2018-01-29:
I just check your matrix.c and I copy these line 

```
    matrix_scan_quantum();  // also missing in original PS2AVRGB implementation

    return 1;
}

void matrix_scan_kb(void) {
  // Looping keyboard code goes here
  // This runs every cycle (a lot)
  matrix_scan_user();
};

```

and its work! I don't even know what it does. Thank you so much again :)

krusli, 2018-01-29:
The QMK docs on [custom functions](https://github.com/qmk/qmk_firmware/blob/master/docs/custom_quantum_functions.md) and [tap dance](https://docs.qmk.fm/feature_tap_dance.html) should be helpful.

In a sense, the original custom matrix scanning code overrides the default matrix scanning code from QMK. It, however, didn't have a call to `matrix_scan_quantum()`, which then calls `matrix_scan_tap_dance()` which tells it to update its internal count/timer (or something along those lines). This is why the tap dance key never seems to timeout and another key has to be pressed to "time it out". Adding that call back fixes it (as it now calls the code for Tap Dance that should run on every matrix scan), but that piece of code expects us to have defined the `matrix_scan_kb()` (keyboard/keyboard directory level) function, which in turn will call `matrix_scan_user()` in the keymap level.

I figured it out while implementing backlighting functions and realising that the backlighting functions, even the built-in ones when not overridden by my custom "driver", meant to be called at every matrix scan did not get called.

underpk, 2018-01-29:
Yes the RGB underglow still works but I rarely turn them on.

danielo515, 2018-01-29:
Hello @krusli 

> meant to be called at every matrix scan did not get called.

May I ask you how did you realized ? Did you pereform any debugging? If so,would you mind to point me to embedded hardware debugging resources ?

Regards

krusli, 2018-01-29:
It was just me reading the code manually and just searching for the function definitions and where they get called. QMK has a built in way to debug using hid_listen (https://docs.qmk.fm/faq_debug.html) but last I remember it wasn't working on the 32A or at least the PS2AVRGB boards.

I did use `send_string()` to just send strings for debugging though although putting it in matrix scanning code is usually too slow and the print gets interrupted.

danielo515, 2018-01-30:
Thank you for your answer.

> I did use send_string() to just send strings for debugging though it didn't work well with longer strings.

Yes,I know. It's very sad

krusli, 2018-01-30:
https://github.com/tmk/tmk_keyboard/issues/12 - with VUSB (the protocol used by the PS2AVRGB boards) Hasu noted that debug messages should go via UART for VUSB (which then can be seen by listening to a serial port).

In JJ40 (and the other PS2AVRGB boards') `rules.mk`:
```C
# unsupported features for now
NO_UART = yes
NO_SUSPEND_POWER_DOWN = yes
```

I tried setting `NO_UART` to `no` along with the other changes for console output but it seems like it results in a lot of undefined variables (maybe it's because of the 32A?)

Maybe implementing something like this (http://binaryupdates.com/usart-in-avr-atmega32a-microcontroller/) and then cleaning it up and moving it into QMK's built-in console/debug implementation could work?

closed 2018-02-02 by underpk