qmk/qmk_firmware #2278: Mod-Tap keys register as tap when used too fast

blastrock, 2018-01-16:
I have a key mapped with CTL_T(KC_SLASH), but if I type ctrl+c too fast for example, the keyboard outputs "/c". The events are probably like:

- press /
- press c
- release /
- release c

I thought that PERMISSIVE_HOLD or IGNORE_MOD_TAP_INTERRUPT was what fixed this behavior, and tried all combinations of them without success. Did I do something wrong, or is there another flag to do what I want?

Taudris, 2018-01-25:
I don't know how you attempted to use these settings, but in case you did something else, you have to put them in your keyboard or keymap's `config.h`. This should cause every .c file to be rebuilt; if it does not, you could be doing your build incorrectly. (This is a shot in the dark; I apologize if you already know all this and have it otherwise working.)

Another thing to check for is potentially conflicting settings and `#undef` statements in files related to your keyboard and layout. That's getting pretty desperate, though.

If you do get these settings to work, beware that they allow the opposite scenario to happen, where you mean to type `/c` but instead get `ctrl-c`. I ultimately found this kind of issue too bothersome and instead dedicated a pinky key to a momentary layer which turns `c` into `ctrl-c` (along with other shortcuts and symbols, such as `{` and `}`).

blastrock, 2018-01-25:
Thanks for the answer!

I did put these settings in a config.h file beside my keymap.c file, and it does trigger a full rebuild every time I change it.

I couldn't find any #undef of these settings in the other files either.

I am aware that it could make the opposite scenario bothersome too, but I wanted to try. This key holding feature is one of the things I like the most about this firmware, I really would prefer to keep it. I will try reducing the TAPPING_TERM delay.

drashna, 2018-03-25:
I find that the `PERMISSIVE_HOLD` causes more issues. Try removing and undefining it, and see if that helps.  And try lowering your `TAPPING_TERM`

fredizzimo, 2018-04-01:
I have been adding unit tests for the tapping system and `PERMISSIVE_HOLD` should do what you want, as long as you don't have `IGNORE_MOD_TAP_INTERRUPT` defined. You can add a `#undef IGNORE_MOD_TAP_INTERRUPT` to you `config.h` file to make sure. Some keyboards define that by default.

From my tests, the options mean the following
* `PERMISSIVE_HOLD` = Always the modified output
* `IGNORE_MOD_TAP_INTERRUPT` = Always the both keys, unless you hold both keys for at least `TAPPING_TERM`
* `PERMISSIVE_HOLD + IGNORE_MOD_TAP_INTERRUPT` = Modified output if the other key is released before the tap key, or if both keys are held longer than `TAPPING_TERM`. So both keys are output when releasing the tap key before the other key and held lesss than `TAPPING_TERM`

That means that the pull request #2325, is proably not needed.

blastrock, 2018-04-15:
Wow I just tried it and it seems you are right. I don't think I tried #undef-ing IGNORE_MOD_TAP_INTERRUPT and I assumed it wasn't defined by default, hence my confusion.

Thank you for sorting this out :)

closed 2018-04-15 by blastrock

der-Daniel, 2019-08-23:
Wow, this has solved my problem, too. :tada: