qmk/qmk_firmware #2279: Is it possible to keep the input from the previous layer when switching?

Miodec, 2018-01-17:
Its pretty hard to describe. Lets say im playing a game, and holding W. If i switch to a layer while holding w, is it possible to lock that input until i come back to the same layer? I tried doing it with kc_trans but that did not work either. My poker worked like that, and i was wondering if its possible with qmk.

drashna, 2018-03-25:
This is by design.  When switching layers, all pressed keys (that aren't mods) are cleared.  

You can see this here: https://github.com/drashna/qmk_firmware/blob/master/tmk_core/common/action_layer.c#L121

But as the comment indicates, this may not be a good idea, as keys can (would) get stuck.  Specifically, if the same key isn't on the destination layer and you let go of it ... it would get stuck pressed.  
Which is why it's handled this way by default.

closed 2018-12-28 by mechmerlin

drashna, 2018-12-28:
This has been changed now.  Update your repo, and it should be the default behavior now.

You can revert by defining `STRICT_LAYER_RELEASE`.

PR #3905 (2018-12-28): Keep pressed keys on layer state change (fixes #2053, #2279)
This allows pressed keys to stay pressed when a layer change occurs.

Say you're playing a game and you're holding W to run forward. You need to access a menu or skill that's bound to F4. You have that set on a momentary layer, so you try pressing Fn+4 — but as soon as you press Fn, QMK resets the W key and you stop moving forward, even though you're still holding the key.
For a visual explanation of the issue, see [this video](https://youtu.be/O4AKOX7F5SA?t=4h30m34s).

This fixes the problem by not resetting held keys when the layer [... cut]