qmk/qmk_firmware #2280: likely bugs in quantum/matrix.c#matrix_is_on and quantum/process_keycode/process_chording.c#process_chording

jjowdy, 2018-01-17:
I'm still working on getting my (Mac) environment so I can write and run tests. (I'm running into Xcode/clang issues with incompatible `-Wa,` flags (`-adhlns`) and haven't deciphered the customized makefiles yet. I think I don't understand what toolchain I'm supposed to be using for running tests—has anybody gotten this to work on Mac?—whereas the avr-gcc stuff works fine for compiling my keymap.)

As a result, I can't yet test this with any confidence and haven't really determined whether this code is actually used. (If it is, I'd love some guidance on how I can actually run these tests and wouldn't mind writing some myself. Then I could just do some tests and would submit PRs for the issues below.)

In any case, I've been reading through the code to try to understand the best way to configure my Ergodox EZ, and I spotted two things that don't look like they could possibly be right:

1. Need to use bit shift rather than lt operation

```
diff --git a/quantum/matrix.c b/quantum/matrix.c
index 5337e2626..d93b67353 100644
--- a/quantum/matrix.c
+++ b/quantum/matrix.c
@@ -235,7 +235,7 @@ bool matrix_is_modified(void)
 inline
 bool matrix_is_on(uint8_t row, uint8_t col)
 {
-    return (matrix[row] & ((matrix_row_t)1<col));
+    return (matrix[row] & ((matrix_row_t)1<<col));
 }
```

2. Probably need to register/unregister all keys in the chord instead of doing an early exit after the first key. But I'm not sure yet of the semantics of register immediately followed by unregister, so I may be wrong. 

```
diff --git a/quantum/process_keycode/process_chording.c b/quantum/process_keycod
e/process_chording.c
index 6c6ebe300..0f076181e 100644
--- a/quantum/process_keycode/process_chording.c
+++ b/quantum/process_keycode/process_chording.c
@@ -66,6 +66,8 @@ bool process_chording(uint16_t keycode, keyrecord_t *record) {
           for (uint8_t i = 0; i < chord_key_count; i++) {
             register_code(chord_keys[i]);
             unregister_code(chord_keys[i]);
+          }
+          if (chord_key_count > 0) {
             return false;
           }
         }
```

jjowdy, 2018-01-29:
I'm pretty sure this code isn't actually used (though fixing it for purposes of having good examples might be worthwhile), so I'll close this out. 
(And fwiw, I did get the tests to run on Mac with this recipe:
```
export PATH=/usr/local/Cellar/gcc/7.2.0_1/bin:$PATH
make test:all AUTOGEN=false CC=gcc-7
# also, symlink from <project_dir>lib/googletest to a local googletest install
```
But that would obviously need some tweaking for different versions.

closed 2018-01-29 by jjowdy

747, 2019-01-27:
Hi, I'm afraid that at least the first piece of code (`quantum/matrix.c`) is still alive and working on my planck light, in a buggy way.

I ran into the bug when I tried to get the position of pressed buttons, which you can reproduce with the snippet below somewhere in your `keymap.c`:

```
  for (uint8_t i = 0; i < matrix_rows(); i++) {
    for (uint8_t j = 0; j < matrix_cols(); j++) {
      if (matrix_is_on(i, j)) { uprintf("WRONG: %d-%d\n", i, j); }
      if (matrix_get_row(i) & ((matrix_row_t) 1 << j)) { uprintf("RIGHT: %d-%d\n", i, j); }
    }
  }
```

Due to the logic failure, `matrix_is_on()` reports for whole row iff you press the leftmost (or index 0) key on the row, while the correct one for the very key you are pressing.

The compiler doesn't warn that it's deprecated, so I assume the function is fair and square, and the behavior should be fixed. Let me know if I'm wrong.