qmk/qmk_firmware #2298: Documentation improvements wishlist

skullydazed, 2018-01-22:
We've been talking a lot lately about what further improvement to documentation we want to make. This is a non-exaustive list, it's mainly so I have somewhere to track this stuff until I have time to tackle it. If you would like to work on something listed below please feel free!

* Environment setup
 * Explain how to clone qmk_firmware
   * Windows
   * Mac
   * Linux
* HOWTO/Guides to Write
 * Update `info.json` files (converter, etc)
 * Code a completely custom key
* How To Do Common Tasks
 * How to clear mods
 * How to send key down, key up

joedevivo, 2018-01-22:
Keyboard development for ARM / ChibiOS? :D

DanielGGordon, 2018-01-22:
Some clarity on Permissive hold and ignore mod tap interrupt - and how they work together.

lbibass, 2018-02-13:
Talk more about Tap Dancing: the documentation wasn't very clear to me, and I feel like some improvements could be made.

evantravers, 2019-04-19:
Where do we stand on this now… I made a pass through documentation a while back… but I'd like to help out again.

drashna, 2019-04-19:
I think that everything but the ARM/ChibiOS stuff exists in the docs.  
However, some of this could be cleaned up.

drashna, 2019-11-21:
Hmm, I think most, if not all of this exists now.

jackhumbert, 2020-03-17:
Yeah, I think we've implemented a good portion of this - we can tackle any outliers in another round :)

closed 2020-03-17 by jackhumbert