qmk/qmk_firmware #2322: AUDIO_ENABLE does not work on Satan GH60

raymond-liang, 2018-01-31:
Audio relies on pins C6 or B5 which I believe are used for the keyboard matrix.

Enabling audio causes the keys on the 5th column to not register or to register all 4 keys at once (4rfc).

Enabling B6_AUDIO (backlight pin) and disabling the backlighting option also does not work and causes the board to be unrecognised.
[https://github.com/qmk/qmk_firmware/tree/b6_b7_audio](url)

Is it still possible to have a speaker on this board?

loadedsith, 2018-03-01:
Can you share a high res pic of the back of the board? I dont know much about this board, but I know on the Planck you want to solder to the promicro's pc6, not the c6 resistor on the pcb...

drashna, 2018-03-03:
The problem here is that most of the pins are already in use. C6, B5, B6 and B7 are all being used by the keyboard matrix. 
So enabling the audio will disable that column. 

So, no, there isn't a good option here.  

You can see a list of the used pins here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/gh60/config.h#L49-L50