qmk/qmk_firmware #2323: I can't find why the linker can't link (I'm poor at programming and English)

MacDamien, 2018-01-31:
I try to create a keymap.c that fit with QMK.

I’m on MacOS High Sierra.

[keymap.c.zip](https://github.com/qmk/qmk_firmware/files/1682513/keymap.c.zip)

what’s my terminal returns, a problem encountered by the linker… :

MBP-de-dv1-2:qmk_firmware dv$ make ergodox_ez:bepo_plover
QMK Firmware 0.5.215
Making ergodox_ez with keymap bepo_plover

avr-gcc (GCC) 7.3.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_ez/keymaps/bepo_plover/keymap.c                                        [OK]
Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/ergodox_ez_bepo_plover.elf                                                          [ERRORS]
 |
 | .build/obj_ergodox_ez_bepo_plover/quantum/process_keycode/process_tap_dance.o: In function `process_tap_dance_action_on_reset':
 | /Users/dv/GitHub/qmk_firmware/quantum/process_keycode/process_tap_dance.c:89: undefined reference to `tap_dance_actions'
 | /Users/dv/GitHub/qmk_firmware/quantum/process_keycode/process_tap_dance.c:89: undefined reference to `tap_dance_actions'
 | /Users/dv/GitHub/qmk_firmware/quantum/process_keycode/process_tap_dance.c:89: undefined reference to `tap_dance_actions'
 | /Users/dv/GitHub/qmk_firmware/quantum/process_keycode/process_tap_dance.c:89: undefined reference to `tap_dance_actions'
 | /Users/dv/GitHub/qmk_firmware/quantum/process_keycode/process_tap_dance.c:90: undefined reference to `tap_dance_actions'
 | .build/obj_ergodox_ez_bepo_plover/quantum/process_keycode/process_tap_dance.o:/Users/dv/GitHub/qmk_firmware/quantum/process_keycode/process_tap_dance.c:90: more undefined references to `tap_dance_actions' follow
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [.build/ergodox_ez_bepo_plover.elf] Error 1
make: *** [ergodox_ez:bepo_plover] Error 1
Make finished with errors

yiancar, 2018-02-09:
Maybe you need to #include "action_layer.h" ?

MacDamien, 2018-02-09:
I resolved this issue, but I forgot to come back here, and now, I don't remember how I solved it. Thank you for your answers, but now I couldn't know if your solutions works.

closed 2018-02-09 by MacDamien