qmk/qmk_firmware #2344: This device cannot start (Code 10) - Plover Steno layout

millipz, 2018-02-05:
[image]
My Plover layout reports this error when connected to my Windows 7 PC. Works as expected on Mac. No keypresses work in Plover. Any suggestions of how to troubleshoot gratefully received.

millipz, 2018-03-26:
Nope, the issue persists. I can only use my steno layer on Mac for now.

drashna, 2018-03-26:
hmmm.  

My best recommendation is to manually uninstall the drivers, and then install the QMK Toolbox (as that should prompt to install the drivers). Or re-run the QMK build enviroment setup script, to force install all the drivers again. 

If that doesn't help, ... then sorry. :(

millipz, 2018-03-26:
I only have QMK toolbox as a standalone .exe. Should I be running an installer? I flashed the keyboard and did all the main setup on my Mac, just downloaded QMK Toolbox to the PC to see if it helped.

millipz, 2018-03-28:
Which drivers should I be uninstalling? I tried removing the driver for the HID device with the issue, but I don't get any driver prompts when installing QMK.

millipz, 2018-05-01:
I finally have this working! I'm writing this in steno!

Steps to fix:

1. Download Zadig http://zadig.akeo.ie/

2. Open and Let's Split should show up

[image]

3. Install USB Serial (CDC) driver

4. Relaunch Plover

5. Profit

closed 2018-05-01 by millipz