qmk/qmk_firmware #2346: new keyboard

Legonut, 2018-02-06:
Hi, I've got firmware for a new keyboard
https://legonut.gitbooks.io/zen-keyboard/content/
I've got ~60 people who will need the firmware, so merging it with QMK master seems like the best plan. Do I need permission to make a pull request?

jackhumbert, 2018-02-06:
Awesome! No permission needed - just go ahead! [We do have guidelines](https://docs.qmk.fm/hardware_keyboard_guidelines.html) :)

Legonut, 2018-02-06:
Thanks, I'm trying to do it through the github windows client, and when I make a pull request it says I don't have permission to publish

Legonut, 2018-02-06:
Ah I needed to fork it, trying again now

drashna, 2018-02-06:
Yeah, fork it, push to the fork and then create a pull request. :)

closed 2018-02-07 by jackhumbert