qmk/qmk_firmware #2348: Levinson - Extra-data Kills right hand keys

mgarcia-accesso, 2018-02-06:
Adding RGB underglow, when attaching left-hand DO from LED strip to extra data pin to trrs cable allows for rgb on right hand but typing breaks. I can plug in the right hand alone and typing is fine. Removing the connection to extra data removes the rgb control on right side but typing returns.

drashna, 2018-02-06:
Are you using I2C or Serial? 

RGB Underglow only works with serial, not I2C.

mgarcia-accesso, 2018-02-06:
#define USE_SERIAL is set in my config.h

drashna, 2018-02-06:
Ah, yeah, the Levingson defaults to using "I2C".  This requires both data channels of the TRRS cable, so the "Extra data" port isn't available.  And this is why it's breaking the keyboard. 

In your keymap folder (where your "keymap.c" file is), create a "rules.mk" file.  If this already exists, then just edit it. 
```
USE_I2C = no
USE_SERIAL = yes
```

Save the file, and then compile.  This should allow you to use both halves *and* RGB.

mgarcia-accesso, 2018-02-06:
P.S. I ran make levinson/rev2:winninggeneration:clean first.

drashna, 2018-02-06:
checking it more .... yeah, there isn't a good solution here.  The backlight uses I2C, so that may be why you're having issues with it. 

And there isn't a good solution here, if that is the case. :(

mgarcia-accesso, 2018-02-06:
@nooges  Seeing the same as above when flashing your keymap bakingpy2u. Did you encounter anything similar?

mgarcia-accesso, 2018-02-07:
Convinced after much testing that it's an issue with an underlying solder point on a trrs jack or something similar. Closing issue.

closed 2018-02-07 by mgarcia-accesso