qmk/qmk_firmware #2353: Debug messages on matrix_init_kb() and pointing_device_init()

raycewest, 2018-02-06:
Trying to get an ADNS9800 working alongside my matrix on a Pro Micro - might elaborate on that in a separate issue, but for here:  can i expect to the init functions to output any print() messages I have in there, or do they run too early in boot process for hid_listen to pick up on them?  I'm not sure if the firmware for the ADNS is getting loaded or not, when the keyboard gets plugged in.

raycewest, 2018-02-24:
If I could follow up:  has anyone else observed this (lack of) behavior?  I can notice a change of a static variable inside of pointing_device_init(), but nothing I can print off.

skullydazed, 2018-03-03:
Have you tried QMK Toolbox instead of hid_listen? I've found that hid_listen doesn't always connect in time, but QMK Toolbox connects in time to catch anything printed after "Keyboard Start". I don't know if the init functions run before or after that.

drashna, 2018-03-25:
Did you get this resolved?

raycewest, 2018-03-27:
No, although I'm no longer looking at this matter to address the problems I'm working through.  Thanks!

drashna, 2018-03-27:
Would you be okay with us closing this issue?

raycewest, 2018-03-27:
Yes, thanks.

closed 2018-03-27 by raycewest