qmk/qmk_firmware #2354: DZ60 default keymappings are not the same as DZ60 rev2

kyleshrader, 2018-02-07:
I just got a DZ60 rev2 usb-c and I have noticed a lot of the keymappings that come with the default firmware do not match up with what is in qmk_firmware/keyboards/dz60/keymaps/default/keymap.c

I assume what was shipped with rev1 and rev2 are different. I can upload a default-rev2/keymap.c if there is a way to extract it, before I flash my own on top of it.

jackhumbert, 2018-02-07:
Unfortunately it's up to the manufacturer/designer to maintain their default keymaps here, and we try to encourage them to do so, along with tagging the release ([our guidelines](http://qmk.fm/powered/) detail this) - if you wanna write them a note about this encouraging them to update it, that would be great!

kyleshrader, 2018-02-07:
I also notice the board is reporting the following:

```
{
    vendorId: 65261, // 0xFEED
    productId: 24672, // 0x6060
    path: '/dev/hidraw8',
    serialNumber: '',
    manufacturer: 'qmkbuilder',
    product: 'keyboard',
    release: 1,
    interface: 3,
    usagePage: 13666, // 0x3562
    usage: 13615 // 0x352F
}
```

Which seems like a different product id than was recorded in the DZ60. In-fact it's the plank's product id. Also the manufacturer and product are different. T_T okay I'll let them know.

closed 2018-02-07 by kyleshrader

jackhumbert, 2018-02-07:
Yeah, I think they use their kbfirmware builder to generate the firmware, and don't update/push things here. We very much wish they would submit PRs here :/ 

`FEED:6060` was used by the GH60, and lots of keyboards in QMK use it because the Planck did, and the original Planck code was copied from TMK's GH60 port cause Matt3o's Brownfox guide recommended that at the time :)

We're working on a solution for custom VIDs for all keyboards though!

kyleshrader, 2018-02-07:
Well before I dump it I will try to fully replicate it (kinda hard when not all keys report out like fn keys) and post it here or somewhere else. It'll gimme something to do while I wait on keycaps and try to replace a broken neopixel

Nicktho, 2018-02-08:
I mapped mine out with http://qmkeyboard.cn/, my keysettings.json is available here https://gist.github.com/Nicktho/8894c0135f20aa23e24e377741531e56. Not sure if that helps