qmk/qmk_firmware #2359: Support for BBC micro:bit?

chowdhary, 2018-02-08:
I can't find any information on microbit support in qmk. Is it even possible? Based on the chibios readme, the support for Nordic nRF51822 (the application processor) is very limited - but only talks about the BLE capability. The microbit also has the Freescale MKL26Z128VFM4 interface processor - which is used by the Teensy 3.x's. How difficult, if at all possible, would it be to port qmk to the microbit?

Details: [http://tech.microbit.org/hardware/](http://tech.microbit.org/hardware/)

skullydazed, 2018-03-03:
We already support the teensy 3.x's, for example see this: https://github.com/qmk/qmk_firmware/tree/master/keyboards/chibios_test/teensy_lc_onekey

Basically, our support for a particular processor is tied to chibios' support.

drashna, 2018-03-25:
Since this is basically answered, and time has passed...

Closing it.  If needed, reopen it or open a new issue if needed.

closed 2018-03-25 by drashna